PX4/PX4-Autopilot #8938: 'make posix_sitl_default gazebo' exiting with Error 1

varunvp, 2018-02-22:
Hello,
The title says it all. I'm running Gazebo 7.11 and ROS Indigo on 14.04. This is what I get when I run make posix_sitl_default gazebo

```
[3/7] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   thread
--   signals
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
--   chrono
--   atomic
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "2.5.0") 
-- Boost version: 1.54.0
-- Looking for OGRE...
-- Found Ogre Byatis (1.8.1)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   thread
--   timer
--   chrono
--   date_time
--   atomic
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
-- catkin DISABLED
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread;-lpthread (found version "2.5.0") 
Gazebo version: 7.11
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   thread
--   timer
--   chrono
--   date_time
--   atomic
-- Configuring done
-- Generating done
-- Build files have been written to: /home/varun/src/Firmware/build/posix_sitl_default/build_gazebo
[4/7] Performing build step for 'sitl_gazebo'
FAILED: cd /home/varun/src/Firmware/build/posix_sitl_default/build_gazebo && /home/varun/CMake/cmake-3.10.2-Linux-x86_64/bin/cmake --build . && /home/varun/CMake/cmake-3.10.2-Linux-x86_64/bin/cmake -E touch /home/varun/src/Firmware/build/posix_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-build
[1/46] Building CXX object CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o
[2/46] Building CXX object CMakeFiles/rotors_gazebo_mavlink_interface.dir/src/geo_mag_declination.cpp.o
[3/46] Building CXX object CMakeFiles/rotors_gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o
FAILED: /usr/bin/c++  -Dgazebo_lidar_plugin_EXPORTS -I/usr/include/opencv -isystem /usr/include/gazebo-7 -isystem /usr/include/sdformat-4.4 -isystem /usr/include/ignition/math2 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -I/home/varun/src/Firmware/Tools/sitl_gazebo/include -I/home/varun/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/include/gazebo-7/gazebo/msgs -I/usr/include/eigen3 -I. -I/usr/include/eigen3/eigen3 -I/usr/local/include/OGRE -I/usr/local/include/OGRE/Paging -I/home/varun/src/Firmware/mavlink/include -std=c++11 -std=c++11 -Wno-deprecated-declarations -fPIC -MD -MT CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o -MF CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o.d -o CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o -c /home/varun/src/Firmware/Tools/sitl_gazebo/src/gazebo_lidar_plugin.cpp
/home/varun/src/Firmware/Tools/sitl_gazebo/src/gazebo_lidar_plugin.cpp: In member function ‘virtual void gazebo::RayPlugin::OnNewLaserScans()’:
/home/varun/src/Firmware/Tools/sitl_gazebo/src/gazebo_lidar_plugin.cpp:150:65: error: call of overloaded ‘isinf(double&)’ is ambiguous
   if (current_distance < min_distance_ || isinf(current_distance)) {
                                                                 ^
/home/varun/src/Firmware/Tools/sitl_gazebo/src/gazebo_lidar_plugin.cpp:150:65: note: candidates are:
In file included from /usr/include/features.h:374:0,
                 from /usr/include/x86_64-linux-gnu/c++/4.8/bits/os_defines.h:39,
                 from /usr/include/x86_64-linux-gnu/c++/4.8/bits/c++config.h:426,
                 from /usr/include/c++/4.8/string:38,
                 from /usr/include/gazebo-7/gazebo/physics/Actor.hh:20,
                 from /usr/include/gazebo-7/gazebo/physics/physics.hh:2,
                 from /home/varun/src/Firmware/Tools/sitl_gazebo/src/gazebo_lidar_plugin.cpp:22:
/usr/include/x86_64-linux-gnu/bits/mathcalls.h:201:1: note: int isinf(double)
 __MATHDECL_1 (int,isinf,, (_Mdouble_ __value)) __attribute__ ((__const__));
 ^
In file included from /usr/include/c++/4.8/random:38:0,
                 from /usr/include/c++/4.8/bits/stl_algo.h:65,
                 from /usr/include/c++/4.8/algorithm:62,
                 from /usr/include/boost/function/detail/prologue.hpp:13,
                 from /usr/include/boost/function.hpp:24,
                 from /usr/include/gazebo-7/gazebo/physics/Model.hh:28,
                 from /usr/include/gazebo-7/gazebo/physics/Actor.hh:24,
                 from /usr/include/gazebo-7/gazebo/physics/physics.hh:2,
                 from /home/varun/src/Firmware/Tools/sitl_gazebo/src/gazebo_lidar_plugin.cpp:22:
/usr/include/c++/4.8/cmath:608:3: note: constexpr bool std::isinf(long double)
   isinf(long double __x)
   ^
/usr/include/c++/4.8/cmath:604:3: note: constexpr bool std::isinf(double)
   isinf(double __x)
   ^
/usr/include/c++/4.8/cmath:600:3: note: constexpr bool std::isinf(float)
   isinf(float __x)
   ^
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
make: *** [posix_sitl_default] Error 1
```
It always exits at 'Performing configure step for 'sitl_gazebo'. I have added the c++ path to PATH.
I searched the net for similar errors, but I didn't encounter any exiting with Error 1.

ChristophTobler, 2018-02-22:
@varunvp Can you test with that branch: https://github.com/PX4/sitl_gazebo/tree/pr-fix_isinf for the `sitl_gazebo` submodule?
```
cd Tools/sitl_gazebo
git fetch origin
git checkout pr-fix_isinf
```

varunvp, 2018-02-22:
I'll try it right away.

varunvp, 2018-02-23:
Thanks Chris, it works now!

closed 2018-02-23 by ChristophTobler

hamishwillee, 2018-09-18:
@QuimItarte This issue was closed. Suggest you repost a new issue with information about your setup. - what OS you are running on, what installation instructions you followed (generally for Linux I use the [scripts here](http://dev.px4.io/en/setup/dev_env_linux.html).