PX4/PX4-Autopilot #8941: VTOL: Position and Mission Reproducible Stall.

tubeme, 2018-02-22:
Hello Guys,

We stumbled on a problem that we tried everything and cannot solve. Here is the case.

VTOL 1.6.5 version. We fly this version, because it allows is to fully disable the airspeed. and the FW_ARSP_MODE=2 thus no speed information so the transition is performed by timing. In 1.7 implementation there is no way we could disable the speed information There is only two options, Air speed and Ground speed for FW_ARSP_MODE.

As far as we understand in this situation in 1.6.5 the TECS are open loop but they still perform extrapolating speed somehow. 

But there is a problem in our configuration. The VTOL flies perfectly well as an airplane in the manual mode stabilized flight, when we have direct control over the throttle. But in Guided and Autonomous modes the throttle response is totally not adequate. So if we Pitch up the plane, it almost does not provide throttle more than the cruising throttle until we hit the limit for the pitch ~25 degrees. then it kicks with a full throttle. 

https://logs.px4.io/plot_app?log=2ead2730-ec02-486d-b30b-336cdd5d5c14

https://logs.px4.io/plot_app?log=51e24d41-7aa1-4dfc-9ef9-7abf3d71fa19

This way we could easily throttle stall the plane every time. The problem persists in the missions as well.

If we make a mission with a steep climb we stall the plane immediately. 

https://logs.px4.io/plot_app?log=2f7629c6-790e-4b5b-b11f-23f3ee16169e

https://logs.px4.io/plot_app?log=36d43908-6b8a-47fb-bab9-74f9a3c44afe

We cannot understand couple of things. 

The TECS is responsible to keep the balance between the energies. So even without speed information from the gyro, acc and altitude the TECS should be able to see the transfer of energies and apply changes to the throttle. Like for example in our case If we pitch up in Position mode (without airspeed and ground speed) it should apply throttle and scale it properly up to the max throttle when we pitch up to the maximum angle. This is pretty simple from pilots perspective even without looking at the speed reading. 

In Altitude mode we are not dependent on GPS lock. So in situation when we fly in Altitude mode without GPS Lock and without speed information, how is TECS performing its corrections?

If we move to 1.7+ where there is no other option but to use ground speed as reference then there are some questions we cannot answer. If in 1.7 we fly in Position and there is GPS problem what will be the strategy for the TECS to balance the plane? Or in 1.7 if there is no GPS lock what is the strategy for the TECS in Altitude mode, how the tecs get their speed info or is it needed at all?

We tried each and every possible parameter in the TECS and all the parameters related to the throttle. We passed each and every parameter in the Attitude controller in the L1 and TECS. We cannot solve this lack of thrust during pitch up in Position mode in airplane mode. 

May be we are missing something very simple. Please help.

sanderux, 2018-02-22:
You need to set tecs heightrate P and I.
Also, increase your cruise throttle and enable voltage and baro scaling.

closed 2018-02-22 by sanderux

tubeme, 2018-02-22:
Thanks @sanderux There is no I gain for the height rate only P and FF gains.

sanderux, 2018-02-22:
Sorry, FF i mean

tubeme, 2018-03-09:
@sanderux Really sorry to bother you but you are the only one who could help me. I dont want to open a new issue so I write here.

We cannot find a balance between the P and FF gains.

We use  FW_T_HRATE_P = 0.15 to exaggerate the problem. During regular flights we use P = 0.07
PX4 1.6.5
QGC 3.2.5
Without airspeed.
Position, Altitude, Mission mode

With FW_T_HRATE_FF = 0.7 we have a good response in the pitch up, The throttle kicks in early and the VTOL FW is adequate. BUT on pitch down if we hold the pitch down for 3-5 seconds and then center the pitch the plane continues to dive with a pitch down for two seconds before it levels. So very scary thing given we have a higher airspeed due to the descend. 

https://logs.px4.io/plot_app?log=9e8c04c1-3553-41bd-9067-de0b41bfad76

From the LOG it is obvious that the throttle is not scaling smooth from 0-max. We noticed that during pith up the throttle goes to 0.8 given we followed your advice to rise the cruise throttle to 0.7

[image]

100% throttle kicks in if we use full pitch up command (1) If we are at 0.9 pitch up then the max throttle goes to 0.8. We know that the plane can handle the current envelope settings give we use Manual mode and control the throttle.

If we use FW_T_HRATE_FF = 0.2 then we have a perfect response on the pitch down. The moment we release the lever the plane levels and the throttle kicks in in level flight. BUT at a pitch up command we have a disaster. The throttle just does not kick except if we make a full pitch up command (1). So at full pitch up the throttle kicks with a 100% at 0.9 pitch up the throttle just stays at cruising 0.7

https://logs.px4.io/plot_app?log=afa6a688-23fe-4521-ace9-6bf43c380888

[image]

Using FF = 0.4 in the middle gives terrible performance to both the pitch up and pitch down.

Because it is better to have good response on pith down we decided to use FF=0.2 and then lower the FW_P_LIM_MAX = 10 degrees in order to not allow the plane to stall.

The ratio of the pushed is very good so the plane itself can perform very well, we could not understand why this is happening in the auto modes...

Can you please help us understand this TECS mechanism given we do not fly with airspeed sensor.

Thanks a lot in advance, I know that the place is not here for this questions, but in the discuss there is very little going on on the responses.

sanderux, 2018-03-13:
There is work in progress currently to improve TECS behavior for airspeedsensorless
I do not understand why you say 1.7 cannot fly without airspeed sensor. we only removes the "groundspeed" option as it wasnt working anyway, so airspeed-disabled does exactly the same.

tubeme, 2018-03-14:
The problem was that we were running the 1.7 with the 3.2 qgc so there was a little unclear where the problem was. So now we managed to fly it since 4 days with 1.7.3 with transition etc.

The problem with the hrate is the same... no difference between the 1.6.5 and 1.7.3

I will post this case with a airplane tag and put some thoughts about solution. This is not something we could fix in any way except with the workarround I mentioned. But this remains workarround and not a fix.

Thanks a lot @sanderux

semihyildirim, 2019-10-24:
Hi,
I have a same problem. I fly lots of time and the plane loss of altitude after transition ends (i.e. begining of fw mode). The height increase during transition in every flight(I can not solve the problem) since height increased during transition, after transition(starts fw mode), the plane want to descent to altitude setpoint(doing by minimum pitching down and minimum throttle) but do this agressively. 
In last fly, I decided to make transition on higher speed so the plane had more speed then previous flights. Changing transition speed to a higher value caused deepstall when plane try to decrease its altitude and I had a big crash. My plane mtow is 25kg. there was no any survided part(including electronics parts) on the plane. It hit to ground with 157 km/h. 

I have opened a issue on: https://github.com/PX4/Firmware/issues/13252

I think that this behaviour executed because of altitude error.
According to me, the plane may not be crashed if the following parameters are set;
*If the height of the plane was higher(tecs can correct the altitude at higher altitudes) but it is not a solution. I do not want to loss of altitude after transition
*Minimum throttle value was more lower value(the plane could not accelerate towards to ground too much when pitching down commanded)
*height rate ff was more lower( the altitude error ignored and max pitching down was not commanded) anyway it may not work
*transition airspeed was more slower(it is tested in previous flights and works)
*minimum pitch value was more higher(pitching down movement was more smooth)

Do you agree with me? or I have a wrong idea about this behaviour