PX4/PX4-Autopilot #8945: Pixracer unable to read data from second GPS

priseborough, 2018-02-22:
**Bug Report**

The firmware at commit 19e706fedefd861d5a975c9379841924c327f668 is unable to read a second GPS receiver connected to the TELEM2 port of a pixracer after making the following modifications:

1) set SYS_COMPANION to zero
2) Modify rcS file to start the second GPS instance, eg https://github.com/priseborough/Firmware/commit/f273157c3ea4ab78ea85a8a12868781264cd4fd5

The following console output is generated when the GPS status is queried:

[image]

Both reeceivers are UBlox M8N units and both work when plugged into the GPS port. The TELEM2 port is able to connect to a modem under normal usage.

priseborough, 2018-02-22:
Based on my testing the upstream firmware support for dual GPS receivers is currently broken. See https://github.com/PX4/Firmware/issues/8945.

I have spoken to Lorenz and expect it to be fixed very soon - they need to for the high reliability/redundant autopilot work that has been started.

priseborough, 2018-03-05:
I have repeated this test with  a pixhawk cube and  it is not possible to get a GPS to connect to any port other than the default GPS port. The test used this commit to prevent GPS or companion computer connections starting on boot:

https://github.com/priseborough/Firmware/commit/8f2faf1d07a58cc3358f8405acc1b4f624e915a7 

On the pixracer I connected the GPS receiver to the TELEM2 port and attempted to start using' gps start -d /dev/ttyS2' 

The driver cycled through a range of GPS protocols and baud rates attempting to connect.

Repeating this test using the GPS connected to the TELEM2 port on a pixhawk 'cube' and starting the GPS via 'gps start -d /dev/ttyS2' produced the same result. I was unable to find a port mapping that worked for the GPS2 port.

bkueng, 2018-03-05:
This should fix it: https://github.com/PX4/Firmware/pull/9019. Can you cross-test it?

closed 2018-03-05 by LorenzMeier

priseborough, 2018-03-05:
Fixed on my hardware TXS.

Rickyg32, 2018-04-01:
Has this issue been fixed in a gold release of code. I have the latest official build on my pixracer and it doesn't see the second GPS so curious how to get it working.

bkueng, 2018-04-03:
Not yet, just in the latest beta version.

Rickyg32, 2018-04-03:
bkueng  thanks man. any idea on timing. I am a month away from wanting to maiden this new build so its not a problem for me right now. But you know what its like,m you want to remove all the bugs as you encounter them before you put it in the air.

bkueng, 2018-04-03:
Hard to say, but a month should be enough. Alternatively you can:
- use the latest beta, we're in a pretty stable state right now
- cherry-pick https://github.com/PX4/Firmware/pull/9019 to stable (lowest risk, if you need it right now)

Rickyg32, 2018-04-03:
Thanks man. I can wait. I have lots to do and the reality is, even if I wanted to maiden tomorrow, missing a gps isn't a real issue. I can wait, I just appreciate your getting back to me. I truly appreciate it

Rickyg32, 2018-04-24:
hows that new release coming