PX4/PX4-Autopilot #8946: Do not RTL while landing

sanderux, 2018-02-23:
When landing in auto mode, the system should not engage RTL on data link loss.

Antiheavy, 2018-02-24:
I agree.  I think this should also apply to low battery failsafe.  There are a few other things to consider:

1) should Return mode still trigger with a Geofence breach when in auto land mode?  Maybe yes?

2) How about loss of RC (and the poorly named "Return to Ground Station" mode)?  Maybe this doesn't count when in an auto land mode...

3) If the Return mode is blocked, there should be an appropriate message logged saying "Return mode prevented, already landing" or something similar.

4) This PR https://github.com/PX4/Firmware/pull/8487 added the new parameter "RTL_LAND_TYPE" ( which will be changed to "RTL_TYPE" if/when https://github.com/PX4/Firmware/pull/8749 is merged).  If RTL_LAND_TYPE (RTL_TYPE) is set to 1 (or 2) then the Return functionality is "fast forwarded" up to the current step in the auto landing sequence.  This is kind of like what you are suggesting here, but critically different because the Return functionality is using a specific pre-planned Land waypoint.  One question would be should these modes be made consistent with whatever gets implemented here in the case RTL_LAND_TYPE (RTL_TYPE) = 0 ?  Specifically, when already in the final auto landing phase (and not prior to), should Return mode be prevented (with appropriate logged message) instead of engaged and "fast forwarded"?

sanderux, 2018-03-01:
@LorenzMeier do you have resources for this?

dagar, 2018-03-01:
This actually overlaps with what I mentioned here. https://github.com/PX4/Firmware/issues/8796

In any mode (LAND, RTL, MISSION) we should know when actually landing. Then we can handle it properly.

sanderux, 2018-03-02:
Should we base this on setpoint_type_land ?

dakejahl, 2018-08-11:
If transitioning from land to RTL then RTL goes directly to the land state. This may not be what everyone else had it mind, current upstream obeys this behavior https://github.com/PX4/Firmware/pull/10044

closed 2019-07-09 by stale[bot]