PX4/PX4-Autopilot #8955: /mavros/vision_pose/pose not showing up in /mavros/local_position/pose

SubMishMar, 2018-02-24:
I am using LPE_FUSION to send vision position and yaw to the px4 FCU but the data in the above mentioned topics are very diffirent. In fact the   /mavros/local_position/pose echoes some constant garbage which doesnt change as  /mavros/vision_pose/pose changes . Any lead on how to fix this issue?

TSC21, 2018-02-25:
@SubMishMar you probably saw that there is an issue template for reporting issues, bugs, etc. Please follow it so we are able to help you.

MulixBF, 2018-03-01:
Hello. 

I am facing similar issue. [Video](https://youtu.be/KX2sJIWfB6U)
I am using default PX4 settings except for:

```SYS_MC_EXT_GROUP: local_position_estimator
ATT_EXT_HDG_M: Vision```

For the reasons beyond my control I have to use old PX4 version:

```pxh> ver all
HW arch: 
SITLFW git-hash: 61120d895b1cfa3798aa6bb2ce4056421ef4de6e
FW version: v1.5.5-2-g61120d8 (1.5.5 0), 17106176
OS version:  (255)
Build datetime: Nov 15 2017 23:24:39
Toolchain: 5.4.0 20160609
UNKNOWN MCU
UID: 0:1:2```

I am running SITL simulation in gazebo on Ubuntu Linux 16.04 with ROS kinetic installed from official repoes.

vuanhduy, 2018-03-06:
Hi @MulixBF,

How do you feed the mocap information into your gazebo?

mhkabir, 2018-04-04:
Closing as no information provided.

closed 2018-04-04 by mhkabir