PX4/PX4-Autopilot #8956: Failsafe enabled: no global position

mathias-h, 2018-02-24:
we are having a problem with our multicoper

it is a Tarrot 650 ironman, with 17" props and a Pixhawk 1 with px4 software.

most of the time it is flying just fine. But then all of a sudden, it it will go completely crazy and freak out. as you can see in the logs provided at 1:38, it drops about 10m meters out of the air. and then recovers.

it gave a no global position error and [lib__ecl] EKF GPS fusion timeout - reset to GPS, and started beeping, so we think there may be a problem with the GPS

hope someone can help.

here is the log: https://logs.px4.io/plot_app?log=f4cc8c37-068c-4537-91ae-43313a89bd04
and here is another one just after where the same error is given: https://logs.px4.io/plot_app?log=e96bd429-0260-476c-a6ec-d7ff9ded233e

Antiheavy, 2018-02-25:
I'm not an expert, but it looks like you have a lot of vibration on your platform:
[image]
Also, you are only tracking 10 GPS sats, depending on what GPS receiver you are using this might be lower than typical.

JonReacher, 2018-02-25:
I second Antiheavy's pointing out of the vibration. Your global position issues coincide with the worst period of vibration.
 We ran into a vibration issue last year because of a poor choice of autopilot mounting foam (although on a FW vehicle). The same issues came up until we changed the mounting foam.

LorenzMeier, 2018-02-26:
I'm closing this - please report back if you continue to have issues after rebuilding the mounting. You should only see one "yellow" area in the plot, corresponding to your propeller RPM.

closed 2018-02-26 by LorenzMeier

mathias-h, 2018-03-08:
now we have installed the new mounting foam, and here are the results, is a lot better, but there are still some vibrations around 40hz, we think look a little wired.
https://logs.px4.io/plot_app?log=f6e60660-c320-400c-95d4-0ebb39d36597

LorenzMeier, 2018-03-08:
Your GPS trajectory looks very weird - what kind of GPS are you flying?

LorenzMeier, 2018-03-08:
Having vibrations at half the prop frequency is not optimal but not totally unexpected either.

mathias-h, 2018-03-11:
we are using the ublox neo-7m gps (https://www.unmannedtechshop.co.uk/ublox-neo-7m-gps-with-compass/)