PX4/PX4-Autopilot #8961: Quadplane tailsitter

LorenzMeier, 2018-02-25:
Fix this: http://discuss.px4.io/t/vtol-quadplane-tailsitter-cannot-finish-transition-phase/5110

RomanBapst, 2018-03-01:
Comments:

- airspeed mode parameter was set to use groundspeed as airspeed, validate that this works
- check if the transition bug in manual mode #8990 is not responsible for this
- elevon actuation is present in transition but very little. Is this because we feed the control signal from the motors to the elevons as well?

dagar, 2018-03-01:
That flight predates the vtol poll changes. https://github.com/PX4/Firmware/blob/3098c77c9168553c5085636fc4b0428e78568bc3/src/modules/vtol_att_control/vtol_att_control_main.cpp#L692

PR #9236 (2018-04-05): tailsitter: do not condition transition on airspeed if airspeed disabled
Should fix #8961
closed 2018-04-05 by LorenzMeier