PX4/PX4-Autopilot #8962: The filter for the D-term should be enabled by default

LorenzMeier, 2018-02-25:
@bkueng We should find a way to enable your improvement by default.

bkueng, 2018-02-26:
Agreed. I thought we should first do the driver sampling refactoring (https://github.com/PX4/Firmware/pull/8731), before discussing filter defaults. @RomanBapst also wanted to look into the whole filtering pipeline.

LorenzMeier, 2018-03-08:
@bkueng What is the status now (10 days later)? Given the time that has passed, is doing more effort realistic or should we get a sane default in? I believe we can choose a filter cutoff that is far from the dynamic limits of pretty much any vehicle and will not impair flight performance.

bkueng, 2018-03-08:
Personally I'm not yet confident enough to say which defaults make sense. I wanted to do some more tests on different vehicles (like the aero fc), which I can do next time I have one available (within the next week or so).
@MaEtUgR also wanted to do some testing on his vehicle(s).

closed 2018-03-13 by LorenzMeier