PX4/PX4-Autopilot #8963: Port mpu9250 to linux

LorenzMeier, 2018-02-25:
@dagar still working on this?

dagar, 2018-02-25:
Not actively, but the idea is to keep pushing enough into `drivers/device` that the final "porting" will be close to trivial. Find and replace a few key things with platform agnostic versions, drop portions now handled by the base class, drop nuttx specific headers.

If there's an immediate need for this I suspect we could get it done in an afternoon.

mcsauder, 2018-05-16:
No reason to wait... This is a positive direction... Let's get to work...  Temp fix merge requests already in queue... additional merge requests to follow... :)

dagar, 2018-05-23:
Initial pieces here. https://github.com/PX4/Firmware/tree/pr-mpu9250

closed 2019-02-09 by stale[bot]