PX4/PX4-Autopilot #8968: When hovering for some time, the drone start to rock then crash.

ToppingXu, 2018-02-27:
when testing, we hover the drone for a short time and the drone start to rock and the pilot changed to stabilized flight mode and haven't given a enough  throttle which made the drone crashed. And the log shows there is error between gps velocity and local velocity. here is the log link.
https://logs.px4.io/plot_app?log=a4aa9f6c-66ee-4d3c-a896-558b521a7ed2

@priseborough can you help analyze this log. Thanks

huangwen0907, 2018-02-27:
you have a bad acc bias

priseborough, 2018-02-27:
There are two possible explanations:

1) There was a yaw estimation error that caused the position controller to start a slow circular oscillation which is also known as 'toilet bowling'. This type of yaw error is normally caused by a misaligned or poorly calibrated magnetometer or by excessive levels of magnetic interference.

Looking at the log the measured magnetic field vector length variation is correlated with vehicle orientation but not with throttle, indicating that calibration, not electromagnetic interference is a possible cause.

mag field strength and roll,pitch,yaw
[image]

2) Errors in accel biases or accel bias estimation caused a coupling between roll/pitch motion and horizontal velocity error that destabilised the position controller. Prior to the  crash the accel bias values for X and Y components were sitting on limits. 

The persistent error in vertical innovations present from the start of logging (see below) combined with the estimation Z accel bias and actual Z accel raedings which are very close to 1g suggests that bias estimation error is more likely than actual accel bias error.

vertical innovations
[image]

There was a bug present in earlier version of the ecl/EKF that could cause accel bias estimates to slowly diverge and hit limits during preflight so this could be the cause if you are using an earlier version of the estimator. 

To help resolve this I require the following:

1) Can you please let me know which version of code was being flown and what version of the ecl library was being used. I was unable to find the commit for the branch flown in the Yuneec repository. 

2) If you remember, can you please tell me which direction the drone was pointing at takeoff. The EKF and magnetomer thought it was 6 deg to the left of true North. Alternatively if the airframe is sufficiently intact and can be powered, are you able to check the mag  calibration by starting the autopilot and checking yaw accuracy outdoors.

ToppingXu, 2018-02-27:
@priseborough thanks a lot for your quick response. The initial heading is pointed  close to north. And I have check the yaw accuracy is good. And I think it is caused by the big acc bias. Later I will check the commit number and let you know.

MaEtUgR, 2018-02-28:
@ToppingXu About the crash because of too low manual thrust:
Please make sure the parameter `MPC_THR_HOVER` is set to a collective thrust value in which your vehicle with average payload is in about in the balance between accelerating up and accelerating down in level hover flight. You can get the first guess for example from a log.
- It scales the upper and lower half of the throttle stick input such that half of the stick input is hover thrust
- It improves altitude controlled (and hence position controlled) flight because the integrator doesn't have to compensate for this offset
- It is necessary for useful land detection because the detector can infer with how low of a thrust the vehicle should loose altitude but it doesn't when it's landed.

ToppingXu, 2018-03-01:
thanks for your help, I have updated the newest ecl library, and doing more test.

closed 2018-03-01 by ToppingXu