PX4/PX4-Autopilot #8969: VTOL: Could not transition with Ground Speed.

tubeme, 2018-02-27:
Hello guys,

We could not make a transition without airspeed sensor but with FW_ARSP_MODE=1

It is valid for 1.6.5 and 1.7.3

Everything in the VTOL section is tuned correctly in terms of airspeed parameters and timing.

The VTOL gains 14ms ground speed which is beyond VT_ARSP_TRANS=12 ms but the quad never stops.

https://logs.px4.io/plot_app?log=ae05e7a8-279f-4b24-9bc3-4ff88962008e

https://logs.px4.io/plot_app?log=02b2b000-0d0d-42bd-b246-496f27af8fce

https://logs.px4.io/plot_app?log=1654e1f9-7ff3-4ec6-8e46-ff6114922485

sanderux, 2018-03-01:
I think groundspeed mode never worked and it has been removed.
In the latest QGC you will not see that option anymore

closed 2018-03-01 by sanderux

tubeme, 2018-03-02:
Thanks @sanderux I'm confused... There were three options for FW_ARSP_MODE 0 - Airspeed, 1 - Ground Speed, and 2 - None in the px4 1.6.5 and QGC 3.2

In PX4 1.7.3 + QGC 3.3 there are only two options left. 0 - Airspeed and 1 - Ground Speed.

Which one is removed? Do you mean QGC 3.4?

sanderux, 2018-03-02:
You can transition with or without airspeed. Depending on mode. Groundspeed is not relevant anymore