PX4/PX4-Autopilot #8970: VTOL: VT_DWN_PITCH_MAX and VT_FWD_THRUST_SC does not apply.

tubeme, 2018-02-27:
We wanted to try the forward pusher motor for forward movement in MC mode.

But it does not apply somehow.

VT_DWN_PITCH_MAX = 3
VT_FWD_THRUST_SC = 2

https://logs.px4.io/plot_app?log=ffe8140b-0971-427b-beae-77298188587e

VT_DWN_PITCH_MAX = 5 (default)
VT_FWD_THRUST_SC = 2

https://review.px4.io/plot_app?log=45c7061b-cc81-48df-bba0-340bea8723c9

Any ideas?

bresch, 2018-02-27:
@tubeme This feature is only active in position control and mission modes in order to be able to control the vtol in stabilized if there is a problem with the pusher.

tubeme, 2018-02-27:
@bresch Thanks we will use it in Position. It is not logic though because it should be consistent in all modes. What kind of problem with the pusher could be fatal for Stabalized or Altitude? If there is some problem with the pusher the VTOL will just use VT_DWN_PITCH_MAX as its maximum pitch angle or am i wrong?

bresch, 2018-02-28:
@tubeme Yes, it will go to `VT_DWN_PITCH_MAX`, however, if this value is close to zero, it becomes quite scary to fly. I had once that value to zero, had a pusher failure and almost crashed to myself during landing because I forgot that I had no pitch down. That's why I decided to remove it from the manual/stabilized mode.

tubeme, 2018-02-28:
@bresch Yes this was the only scenario I could thing that will be dangerous. If you put "0" :) I thought that it applies to the Altitude as well. But you just removed it from Manual modes... thanks I'm closing this.

closed 2018-02-28 by tubeme

bresch, 2018-02-28:
@tubeme Okay, but if you think that you really need this enabled during altitude mode, just open an issue for a feature request and we could make a modification.