PX4/PX4-Autopilot #8971: prevent RTL indoors

Stifael, 2018-02-27:
In order to do RTL, the vehicle requires a valid home position as well as a valid global position. These requirements, however, can also be satisfied indoors because it just depends on the quality of the gps. RTL can not only be triggered by the user, but also automatically as low battery failsafe action ([COM_LOW_BAT_ACT](https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander_params.c#L297-L309)). 
There are several options how to deal with this rather dangerous trigger when flying indoors. A few possibilities:
- ask the user before triggering rtl
- add another indoor mode that has nothing to do with Altitude mode that is available indoors and outdoors and therefore incorporates all the failsafe actions
- modify the current Altitude mode such that rtl is never triggered

dagar, 2018-02-27:
Commander's validity flags (local position, global position, velocity) depend on configurable thresholds. Can we tighten those requirements?

Do you have a few representative logs on hand we could look at?

dagar, 2018-02-27:
Relevant - https://github.com/PX4/Firmware/pull/8771

LorenzMeier, 2018-02-27:
I also believe that these should be suitable to reliably detect this.

proficnc, 2018-02-27:
Great feature request!

Stifael, 2018-02-28:
log: https://logs.px4.io/plot_app?log=749ca8df-dbed-47f4-bfc5-53c70b2d46bc
video:  https://youtu.be/Ax8xiDSo17w

Sure, threshold adjustment might be good as an intermediate step. We will see if it is reliable or not.

LorenzMeier, 2018-05-27:
We have improved the indoor detection in ECL, so closing this.

closed 2018-05-27 by LorenzMeier