PX4/PX4-Autopilot #8973: hardfault

stmoon, 2018-02-27:
Hello,

I found the hardfault with firmware px4fmu-v3_default (v1.7.3)

Could you tell why the hardfault happens and how to fix it?

I link the ulog (https://logs.px4.io/plot_app?log=71cff15b-4655-4f6b-a35e-e5675c057866)

Thanks in advance

```
[hardfault_log] -- 2000-01-01-01:46:31 Begin Fault Log --
System fault Occurred on: 2000-01-01-01:46:31
 Type:Hard Fault in file:armv7-m/up_hardfault.c at line: 171 running task: mavlink_rcv_if0
 FW git-hash: f293593a19605c0de63d24d2ef1456ce3349e8bc
 Build datetime: Feb 19 2018 10:41:16
 Build url: localhost 
 Processor registers: from 0x20014ec4
 r0:0x100079a0 r1:0x200145a8  r2:0x10004c08  r3:0x20003ab0  r4:0x00000000  r5:0x08360000 r6:0x28f9004a r7:0x20015220
 r8:0x0813c1cc r9:0x00000000 r10:0x0000001c r11:0x00000000 r12:0x000000c8  sp:0x20014f98 lr:0x0800b98f pc:0x000bdace
 xpsr:0x00000000 basepri:0x000000f0 control:0x00000004
 exe return:0xffffffe9
 IRQ stack: 
  top:    0x200029f0
  sp:     0x200029a0 Valid
  bottom: 0x20002704
  size:   0x000002ec
  used:   000000c8
 User stack: 
  top:    0x20015210
  sp:     0x20014f98 Valid
  bottom: 0x200146fc
  size:   0x00000b14
  used:   00000b14
Interrupt sp memory region, stack pointer lies within stack
0x20002a51 0x10006870
0x20002a50 0x000000b7
0x20002a4f 0x20010780
0x20002a4e 0x000000b6
0x20002a4d 0x10007470
0x20002a4c 0x00000075
0x20002a4b 0x1000e190
0x20002a4a 0x00000114
0x20002a49 0x20017520
0x20002a48 0x000000b3
0x20002a47 0x20007fa0
0x20002a46 0x00000112
0x20002a45 0x200176b0
0x20002a44 0x00000071
0x20002a43 0x1000c900
0x20002a42 0x000000d0
0x20002a41 0x20014530
0x20002a40 0x000000cf
0x20002a3f 0x1000e380
0x20002a3e 0x0000ffff
0x20002a3d 0x00000000
0x20002a3c 0x0000ffff
0x20002a3b 0x00000000
0x20002a3a 0x0000ffff
0x20002a39 0x00000000
0x20002a38 0x0000ffff
0x20002a37 0x00000000
0x20002a36 0x0000ffff
0x20002a35 0x00000000
0x20002a34 0x00000109
0x20002a33 0x20015f70
0x20002a32 0x0000ffff
0x20002a31 0x00000000
0x20002a30 0x000000c7
0x20002a2f 0x2000e430
[... 696 lines cut ...]
0x20014eef 0x21000000
0x20014eee 0x08011188
0x20014eed 0x08007373
0x20014eec 0x000000c8
0x20014eeb 0x10004b90
0x20014eea 0x10004c08
0x20014ee9 0x200145a8
0x20014ee8 0x00000002
0x20014ee7 0x00000000
0x20014ee6 0x00000000
0x20014ee5 0x00000000
0x20014ee4 0x00000000
0x20014ee3 0x00000000
[hardfault_log] -- 2000-01-01-01:46:31 END Fault Log --
```

**Bug Report** or **Feature Request**

For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](slack.px4.io). Issue reports need to contain the items below:

 - Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/flight-reporting.html))
 - Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/)
 - Expected behavior and actual behavior.
 - Steps to reproduce the problem.
 - Specifications like the version of the project, operating system, or hardware.

bkueng, 2018-02-27:
Hi

Thanks for the report. Since you have additional code changes, it will be very hard to help you. Can you flash the officially released version (1.7.3) and see if you have the same issue?

Which hardware do you have attached to the board, and to which ports?

There's currently another hardfault report here: https://github.com/PX4/Firmware/issues/8913. It's different that the fault happens in `sem_post.c`, whereas here it's `armv7-m/up_hardfault.c`. But if it's due to memory corruption, it can still be the same cause.

stmoon, 2018-02-27:
Hi, @bkueng 

yes, I modified something  to the firmware. However, I think that it is not related to the problem because it was working well until yesterday.

Maybe, I just replaced GPS. and it is connect to GPS2 (not GPS1). 
GPS1 connector is not used. (it just used for safety button).

I think it is related to the problem.. 

I have used pixhawk 2.1.

closed 2019-02-09 by stale[bot]