PX4/PX4-Autopilot #8980: FW unsuccessful flight 02/26/18

r0gelion, 2018-02-28:
The pilot flew an FW with Pixhawk Mini and master firmware (38e5a06) and faced the next problems:

1: It took a long time to connect to QGC by the telemetry radio, and when it was connected it didn't give data.
2: When it was flying stabilized mode the plane oscillated a lot.
3: In mission mode, the plane was only gaining altitude or advanced to the first waypoint and moved very slow.

The plane only flew well in manual mode.

Logs of the flights:
https://logs.px4.io/plot_app?log=316d5dcf-4773-4298-8e6e-68d85a8d2aeb
https://logs.px4.io/plot_app?log=2e6acbbd-967f-44ed-9ac2-26885cc45720
https://logs.px4.io/plot_app?log=73cdcdbe-128e-41cf-84df-e61ab3bde83d
https://logs.px4.io/plot_app?log=bfc0ef2e-0fdf-417d-b2e3-7f65441837d8
https://logs.px4.io/plot_app?log=29a06e2d-b38b-4a4a-b8f5-417b8f9c50ac
https://logs.px4.io/plot_app?log=6ab92bec-5d20-45b4-a387-fd86e5117a3c

Previous successful flights:
https://logs.px4.io/plot_app?log=e91adbbe-1da4-49a9-a9e3-688921b03f50
https://logs.px4.io/plot_app?log=3005515b-a35e-4cae-8dd2-2fa3176e0c8d
https://logs.px4.io/plot_app?log=ff408fb9-e829-4db4-ab53-c0cbfb9764c6

@dagar @RomanBapst

RomanBapst, 2018-03-01:
@r0gelion Thanks for reporting this, I will look at it tomorrow.

RomanBapst, 2018-03-04:
@r0gelion It looks like the unsatisfactory attitude control was caused by false airspeed measurements.
The airspeed data from the unsatisfactory flight is unrealistically high (30 - 50 m/s) but this vehicle cruises around 15 m/s. As a result the rate controller was scaling down the control surfaces and thus leading to degraded performance.
Notice that at the end the airspeed actually drops to zero and as a result the controller scaled the control surface signals up leading to heavy oscillations.

It would be interesting to find out what happened to the airspeed sensor.
In the future we will want to observe airspeed innovations and neglect airspeed data if they exceed a certain limit. I think in this flight it would have been very obvious that the airspeed data is garbage.
@priseborough FYI

r0gelion, 2018-03-06:
@RomanBapst thank you for the analysis, it makes sense.

We cleaned the pitot tube, reset the parameters and calibrated the sensors again and it flew well.

Here is a log:
https://logs.px4.io/plot_app?log=c4fce3b0-01c6-43f2-bbbb-c3231736e1eb

Regards.

r0gelion, 2018-03-06:
BTW, I noticed a NaN error in Roll and Pitch Angles graphs:
https://logs.px4.io/plot_app?log=c4fce3b0-01c6-43f2-bbbb-c3231736e1eb#Nav-Roll-Angle
https://logs.px4.io/plot_app?log=c4fce3b0-01c6-43f2-bbbb-c3231736e1eb#Nav-Pitch-Angle

However, the pilot didn't report any issue.

cesarengineer, 2018-03-07:
@r0gelion in this case the airspeed sensor required some kind of calibration before flight?

santiago3dr, 2018-03-08:
Few flights on the same rig:
https://logs.px4.io/plot_app?log=b7c5b615-63fe-413a-9ffc-ee4e5117f6d7
https://logs.px4.io/plot_app?log=a5b4bfa3-7c7a-487a-9d9f-13d1930c265a
https://logs.px4.io/plot_app?log=06b303f9-5f13-471e-824d-c8668d9dce0f

flying wing is flying fine now that the airspeed sensor issue has been addressed; everything looks to be working as it should

r0gelion, 2018-03-08:
@cesarengineer we found the pitot tube was pushed back against one of the silicon tubes damaging it. We fixed the issue, cleaned the pitot tube and since we changed some parameters before we knew what was the problem, we reset the parameters and repeated calibrations.

cesarengineer, 2018-03-09:
@r0gelion All right., which parameters you changed after fixing this issue?

cesarengineer, 2018-04-01:
@r0gelion thanks for you response. Just a quick, based on the user's guide, during the calibration of this sensor it is required to blow across it? how it should be done exactly? thanks.

closed 2018-08-01 by Tony3dr