PX4/PX4-Autopilot #8992: Dropix loosing LSM303D after Reboot

tops4u, 2018-03-01:
**Bug Report**

I have two Dropix 2.1 Units which have 3 IMU Sensors. One is a LSM303D connected via SPI Bus. On initial Boot (Power Up) I can see 2 Accelerometers, 2 Gyro, 1 Baro and 1 Mag for command "sensors status". If I order a reboot via shell or from QGC due to Parameter change the LSM303D disappears. 

In the Bootlog it shows as not found. When I try to start it Manually it will complain about not finding the Sensor on the SPI Bus. 

On Firmware 1.6.5 this does not happen for both units.

dagar, 2018-03-01:
I'll have a fix for this shortly.

dagar, 2018-03-07:
Should be fixed by https://github.com/PX4/Firmware/commit/ba90952e071259b5d0293a7b876725f7df3db6a0

tops4u, 2018-03-11:
Fixed. Thanks @Dagar

closed 2018-03-11 by tops4u