PX4/PX4-Autopilot #8998: Bug: Quad+Tailsitter (AutoStart ID 13004) Not work

jingego, 2018-03-02:
the Quad+Tail-sitter can not be found and set in QGC Airframe page, only Quad_X_Tail-sitter is available. 
[image]

No signal on output ports, if I manually set SYS_AUTOSTART to 13004 in Parameter list.

dagar, 2018-03-02:
We need to test all of these configs.

RomanBapst, 2018-03-02:
@dagar the problem is that this config is excluded from the build due to flash space

dagar, 2018-03-02:
@jingego which version of firmware is this? Did you flash through QGC?

@RomanBapst In that case I'm wondering if we need to do a better job forcing QGC metadata to update. I've found the airframes get stuck and don't necessarily update after a flash even after you restart QGC. @DonLakeFlyer @dogmaphobic FYI

jingego, 2018-03-03:
I flashed it through both QGC and make px4fmu-v3_default upload. The version is 1.7.3 dev. I tested two boards but both are hobbyking pixhawk.

jingego, 2018-03-16:
@RomanBapst I saw the [code](https://github.com/PX4/Firmware/blob/1930cc2fbeb9c855f5ae6f99c0ecfd4edabb8b0c/ROMFS/px4fmu_common/init.d/13004_quad%2B_tailsitter#L19). board px4fmu-v2 is excluded. 
it's seems that px4fmu-v3 use same config file with px4fmu-v2, so this airframe is not available on px4fmu-v3 either?
But px4fmu-v3 don't have flash size problem.

closed 2019-02-09 by stale[bot]