PX4/PX4-Autopilot #9000: EKF2_AID_MASK: rotate external vision in 1.7.3

Changliu52, 2018-03-02:
Hello,

Just want to ask for some information about the newly added option rotate_external_vision(6) for EKF2_AID_MASK.
I wonder if I have a SLAM measurement in NED, but not in true-north, do I need to enable this? and automatically getting the position in true-north by fusing the magnetometer? 

If that is the case, will it be problematic if I also enable vision_yaw_fusion (4)? Since from what I understand, enabling vision_yaw_fusion will disable magnetometer fusion. 

If that is not the case, what is the purpose rotate_external_vision option?

Thank you in advance.

Chang

Changliu52, 2018-03-06:
hello?

mhkabir, 2018-03-06:
You cannot enable both fusion flags at once. They are mutually exclusively in the current implementation. Enabling rotate_external_vision will estimate the offset between the vision frame and true NED. The position estimate will be in true NED. The EKF thus keeps using the compass for yaw estimation, and this is not ideal, but we are working on a better solution.

dagar, 2018-03-06:
Do we need an additional metadata mechanism to explicitly block invalid combinations? The other quick thing we could do is put a check in EKF2 updateParams() to log an error.

Changliu52, 2018-03-06:
Thank you for the clarification @mhkabir.
Great! That's one step forward! Despite the magnetic interference, at least we can always fuse GPS + Vision together! (that is what's happening when enables use_GPS + vision_position_fusion + rotate_external_vision, right?)

Great work!

closed 2018-03-18 by Changliu52