PX4/PX4-Autopilot #9002: /mavros/local_position/pose shows nothing after publishing /mavros/vision_pose/pose

vuanhduy, 2018-03-03:
**Bug Report**
My system setting as follows:
- Flight controller: Pixhawk 1
- PX4 firmware: 1.7.3 (installed from qgroundcontroll)
- Onboard Computer: Raspberry Pi 3 installing: Ubuntu Mate + Ros Kinetic + Mavros + Mavros Extras + vrpn-client-ros
- Mocap system: Optitrack

On qgroundcontrol, I set EKF2_AID_MASK to 24 (obtaining position and yaw from mocap system), EKF_HGT_MODE to 3 (vision). On Rasp, I run mavros, vrpn_client_ros and topic_tools to relay the data to /mavros/vision_pose/pose. Finally, I subscribe /mavros/local_position/pose but nothing shows up.  Could you guy please double check this issues?

jingego, 2018-03-03:
I use VICON. as my point of view, the mocap topic should be "/mavros/mocap/pose".

vuanhduy, 2018-03-03:
Hi @jingego,
Thank you for your reply.
The "/mavros/mocap/pose" and "/mavros/vision_pose/pose" topics confuse me, too. To my humble knowledge, there are two ways to work with motion capture system:
1. LPE as shown in [here](https://dev.px4.io/en/ros/external_position_estimation.html). Then the mocap data should be fed to /mavros/mocap/pose topic.
2. EKF2 as shown in [here](https://github.com/PX4/Firmware/issues/8520). Then the mocap data should be fed to /mavros/vision_pose/pose topic.

Our lab has 2 drones: Intel Aero and the one with pixhawk 1 (we built it ourself.) We have tried both approaches on our Intel drone and prefer the later one. In other words, we do exactly configuration for both drones, and only the local position topic (/mavros/local_position/pose) of Intel one shows us the information but not the pixhawk one.

TSC21, 2018-03-03:
> Finally, I subscribe /mavros/local_position/pose but nothing shows up

It will only "show up" something if you actually have a position lock on the vehicle side, meaning, it "knows" where it is. How are you setting your serial link with the Pixhawk (`SYS_COMPANION` value, baud rate on the MAVROS side, etc.)? Are you even sure that you have a proper link setup between the Rapsberry Pi and the Pixhawk? How's the wiring done? Is it on TELEM1 or TELEM2? Using `rostopic echo -n1 /diagnostics` on your Pi, check if you actually have the link setup and you get any data at all.

vuanhduy, 2018-03-03:
Hi @TSC21,
Thank you for your quick reply.
To connect pixhawk with raspberry pi:
1. On pixhawk, we are using TELEM2 and set the `SYS_COMPANION` to 921600.
2. On raspberry pi, we are utilizing an uart-usb adapter (yp-05), adjusted px4.launch file to use `/dev/ttyUSB0:921600`, started mavros by `roslaunch mavros px4.launch` command.

We checked the connection and was able to get data  from /mavros/state, /mavros/imu/mag and /mavros/imu/data_raw by the command `rostopic echo`. We also do echo on /mavros/imu/data topic but nothing shows up.

I do not have the access to the drone now but I will do `rostopic echo -n1 /diagnostics` and let you know as soon as I can.

TSC21, 2018-03-03:
What streams have you configured on the `mavlink_main` on the vehicle side? If are not actually publishing the required msgs, you will never get them on the RasPi side. Also, same question: do you get position lock on the vehicle? (green led fading off and turning on slowly) Do you get these msgs on QGC if you connect the vehicle to it?

jingego, 2018-03-03:
if you have another vehicle actually work, how about ros network setting？ we had similar problem due to ros network setting. I hope my experience could help.
May be you can have a look at `rostopic info /mavros/vision_pose/pose`, check the publishers and subscribers. In our case, if I have one like `http://XXXX-PC:56789` and the other one like `192.168.0.1:67890`, the subscriber sometimes actually do not get any data. we think it's a name resolution problem. This also happened when I use a single machine but using Matlab ROS Toolbox to publish data. To solve this problem, I will set all the hostnames to the IP of the machines.

vuanhduy, 2018-03-05:
Hi @TSC21,
 - [Here](https://drive.google.com/file/d/1adwSlk2aWNOK04XOcOdOfwm5s0suVlw9/view?usp=sharing) is my mavros log.
 - [Here](https://drive.google.com/file/d/1KHrsMyygi6wd_lqxVqV7Gzb696f4kGQ4/view?usp=sharing) is for diagnostic
 - [Here](https://drive.google.com/open?id=1byKMbzSdmcYgAPeH_juJbMFu_5Ygugb3) are what my mocap system published.
 - The BLUE led does fade off and turn on slowly (I cannot see any green led)
 - I did not compile the code for my pixhawk but plugged it into the QGC to update the firmware. So my expectation is that the configuration of `mavlink_main` should be set to default.

TSC21, 2018-03-05:
What configs do you have on `px4_config.yaml` on the launch dir of MAVROS? Did you setup the correct plugin configs for `mocap_pose_estimate`? Also, EKF2 does not support mocap data fusion, you would have to: or use LPE if you want to publish the data as `ATT_POS_MOCAP` or use the `VISION_POSITION_ESTIMATE` msg (`vision_pose_estimate` plugin) so to use it with EKF2. Keep `EKF2_AID_MASK` set to 24.

vuanhduy, 2018-03-05:
Hi @TSC21,
Here are my [px4_config.yaml](https://drive.google.com/open?id=1msdDdiVU3MmIsgw5PPZpzpXGhES9I4kD) and my [px4_pluginlists.yaml](https://drive.google.com/file/d/1-Nq274MjZKgn2ZJaCt6X7D3NZ4NibBae/view?usp=sharing).
Fromt the mavros.log, I can see that my vision_pose_estimate plugin has been loaded and initialized, and I did set `EKF2_AID_MASK` to 24 and `EKF2_HGT_MODE` to 3.
I have a very stupid suspicious is that my controller (pixhawk 1) is too old and is not supported vision_pose_estimate feature?

TSC21, 2018-03-05:
Are you publishing the mocap data to `/mavros/vision_pose/pose` topic at all? Have you tried doing what @jingego said about using `listener vehicle_vision_position 5` with the Mavlink Console of QGC to check whether your FCU is receiving the data or not?

> I have a very stupid suspicious is that my controller (pixhawk 1) is too old and is not supported vision_pose_estimate feature?

`vision_pose_estimate` -> MAVROS plugin (independent from any autopilot)
`VISION_POSITION_ESTIMATE`-> Mavlink msg (that actually `vision_pose_estimate` sends into the serial link) - this msg is supported in PX4 firmware since a long time ago (it will be 4 years this year). So unless you have a very old FW loaded into your Pixhawk, then you should be fine (which is not the case).

vuanhduy, 2018-03-05:
I got an error: "listener: command not found"

TSC21, 2018-03-06:
Yeah forget it. That is only available in `px4fmu-v4` and Linux. You will have to add a probe on the `mavlink_receiver` so to be sure that you at least are getting the data. If you actually get the data, it may also mean that the data you are feeding may have something wrong with it, like the timestamps.

TSC21, 2018-03-06:
I also recommend you try the latest upstream master FW, since it added some substantial changes on EV support for EKF2 that may fix your issue.

vuanhduy, 2018-03-06:
Hi @TSC21 and @jingego 
Thank you for your support. The problem was fixed by the dev firmware (compiling from the source code). I guess that the feature is not finalized so it was disabled in the stable-release version. @TSC21 could you please help me to close this thread.

closed 2018-03-06 by TSC21

sergioma295, 2019-06-07:
I am having a similar problem. I have created a new issue. Can somebody help me please? #1248

julianoes, 2019-06-07:
@sergioma295 please don't necrobump. Open a new issue and describe your problem in detail, thank you.