PX4/PX4-Autopilot #9003: Unexpected position setpoint during loiter mode after pause

JohnSnowball, 2018-03-03:
**Bug Report**
Standard vtol, normal flight.
During RTL, I paused the plane for a few seconds to watch, then I tape on the "land" command, however, it goes straight forward and hits a tree.
I checked log, after the "land" command is sent, an unexpected position setpoint is created, while mission setpoint type is not valid.
Here is the log:
https://logs.px4.io/plot_app?log=4cbc668e-2688-41a0-a09c-8f9758fc70c0 

log screen shot is here:
[image]

JohnSnowball, 2018-03-06:
this confused me a lot, as I have done many times in simulation, but could not reproduce this bug, why in that time, the plane doesn't go into landing flight mode, instead it stayed in loiter mode(repositioning) and flies to an unknown position? 
@julianoes @DrTon  I really need your help.

julianoes, 2018-03-06:
@JohnSnowball thanks for the report. I have to defer VTOL questions to @RomanBapst.

JohnSnowball, 2018-03-22:
I found it comes from an unexpected "rally" point command from QGC.

closed 2018-03-22 by JohnSnowball