PX4/PX4-Autopilot #9004: generic airframes

dagar, 2018-03-03:
We should have a wide range of generic airframes at different sizes. Many of our specific airframes are actually fairly generic, and users rarely have the same thing.

I also don't think we should carry exact configurations for something that isn't well defined. Either available off the shelf or with a very specific build log.
 - https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/13008_QuadRanger
 - https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/13001_caipirinha_vtol
 - https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/13010_claire

MC
 - different sizes for quad
  - 180, 250, 330, 450, etc
  - 
 - drop effectively generic options like flamewheel 330, 450, zmr250
 - variations with vmount and uavcan should be configurable via parameters on top of defaults, rather than new airframes

FW
 - a few common sizes and configurations

VTOL
 - tailsitter
   - duo, tri, quad
 - quadplane
   - quadranger 
 - tiltrotor
   - not sure if there are generics here?

I'd like to see a generic option as the default for each frame type.

[image]
[image]
[image]
[image]

dagar, 2018-03-03:
We should also get rid of this.

[image]

hamishwillee, 2018-03-05:
I think this makes a lot of sense. Will users understand how to map best "generic" case to use for (say) flamewheel 330? 

> variations with vmount and uavcan should be configurable via parameters on top of defaults, rather than new airframes

Any variation that requires setting parameters might be a lot harder for users?

dagar, 2018-03-05:
> Any variation that requires setting parameters might be a lot harder for users?

I was thinking the opposite. It's not like most users wishing to use vmount will have the exact airframe that offers vmount.
Parameters can also be broken out into accessible high level QGC gui configuration.

hamishwillee, 2018-03-05:
I was imagining a scenario where I purchased an RTF DJI flamewheel - previously I could just select an airframe and now I have to additionally set some parameters. Very happy to accept I know nothing about this :-)

philipoe, 2018-04-05:
I also agree the airframe groups need to be changed, at least for the fixed-wing planes. I openened https://github.com/PX4/Firmware/issues/9118 a couple of days ago because I had not seen this discussion.

closed 2019-02-09 by stale[bot]