PX4/PX4-Autopilot #9020: Copter gets armed when EKF2_AID_MASK = 9 and no VPE is provided, EKF2 provides no attitude estimate

shylent, 2018-03-05:
I am running v1.7.3, estimator is EKF2, EK2_AID_MASK is 9 (GPS and vision position).

I've attempted to fly the copter indoors in stabilized mode (there is no GPS fix) without providing VISION_POSITION_ESTIMATE messages. Flight mode changed successfully, the copter was armed and when I've applied throttle, it took off, but was completely uncontrollable with regards to roll and pitch.

Strangely, if you set EKF2_AID_MASK to 1 (GPS only) and have no GPS fix, you can still take off in modes, that don't need GPS fix (altitude control, stabilized) and everything works as intended.

I've expected the copter to either to arm and fly normally or at least refuse to arm.

Here is the flight log: https://review.px4.io/plot_app?log=0d1bf158-5559-4bb6-ab31-c23580292b4a

ChristophTobler, 2018-03-05:
Your vehicle didn't have an attitude. That's why it was uncontrollable. Have to have a closer look why that was.

shylent, 2018-03-05:
> Your vehicle didn't have an attitude

That's what it says in the issue title, yes :)

ChristophTobler, 2018-03-08:
True...
Does it happen every time? Because I can't reproduce it.

Just a general side note: Always check your attitude in QGC before arming your vehicle

shylent, 2018-03-08:
Yes, it happened to me consistently. I have no access to the real vehicle right now, but I've reproduced this behaviour using SITL.

I've also found, that even if you have a GPS fix, but no VPE messages are received, while EKF2_AID_MASK = 9, attitude estimate is not provided as well, which is probably undesired behaviour.

It can be reproduced in v1.7.3 and in current master (2ce8b6a9) as well.

I've recorded flight logs using gazebo for simulation and QGC to arm the vehicle.
I am attaching the patches I've used to reproduce (for some reason github does not allow `.diff` files, so I've uploaded them as `.txt`) - the change in simulator_mavlink.cpp is there so that VISION_POSITION_ESTIMATE messages, that come from gazebo, are ignored by the simulator app.

[v1.7.3_no_gps.txt](https://github.com/PX4/Firmware/files/1794854/v1.7.3_no_gps.txt)
https://review.px4.io/plot_app?log=fcd0355c-c4ac-44c3-b4e7-9d95b0677021

[v1.7.3_with_gps.txt](https://github.com/PX4/Firmware/files/1794855/v1.7.3_with_gps.txt)
https://review.px4.io/plot_app?log=2f196a0f-b51a-4075-bde3-443d13f3fe29

[master@2ce8b6a_with_gps.txt](https://github.com/PX4/Firmware/files/1794857/master.2ce8b6a_with_gps.txt)
https://review.px4.io/plot_app?log=924a798c-aa85-4cae-9516-619f1722c54c

ChristophTobler, 2018-03-09:
I see. So the issue is that the flight controller / EKF2 is not receiving any external vision position messages and therefore the EKF2 is not initializing (same for your original log). This is intended and therefore not a bug.

FYI: in current master you have to use optical flow model to simulate external vision position (`make posix_sitl_default gazebo_iris_opt_flow` and change the `EKF2_AID_MASK`) or add the plugin to the iris model.

mhkabir, 2018-03-10:
I think we should prevent arming when the EKF did not initialize. I believe that this was the case before. @dagar FYI.

shylent, 2018-03-10:
It seems, I can't reopen this issue because non-members can't do that.

mhkabir, 2018-04-04:
@dagar We should work a fix for this into the preflight checks, and deny arming without an attitude estimate. Also potentially modify the `ARM_WO_GPS` parameter to `ARM_WO_POS` (encompassing GPS, VIO, Mocap, etc.), so that it is more generic.

dagar, 2018-04-04:
@mhkabir let's start a new issue to discuss. It's a relatively easy change to make (after https://github.com/PX4/Firmware/pull/9193 goes in), but I'd like to make sure we capture and expand all of these arming requirements consistently.

I also want to consider if we should be determining some of them automatically. For example if you have data link loss configured to RTL you should be required to have a valid position estimate and home position for arming.

closed 2019-02-09 by stale[bot]