PX4/PX4-Autopilot #9022: Throttle spike on arming

Vavooon, 2018-03-05:
In rare cases after arming throttle level is higher for a second, then goes back to normal values.
I noticed it first time when I used 1.7 firmware, however there is similar topic from previous year http://discuss.px4.io/t/slight-higher-idle-throttle-when-first-arm/2471 .
In related issue was mentioned that MPC_THR_MIN could affect but after I lowered the parameter nothing has changed.
Here's the log https://logs.px4.io/plot_app?log=7c0b159c-bf28-48dc-8677-b3ccd1c2d25b1 .

I've also noticed that it happens after first arm.

michael1874888, 2018-03-13:
Have you solved the problem?
We have faced the same problem recently.

Vavooon, 2018-03-13:
@michael1874888 Not yet, just waiting for some response.
How high is throttle level in your case? Is it close to hover or just a bit bigger than idle level?

michael1874888, 2018-03-14:
Our drone would fly very high at the beginning, and it became normal in few seconds.
We thought that it is cause by the firmware version, it might be a bug in the newest version.
After we degrade to PX4 firmware to 1.6.5 and it became normal.

RomanBapst, 2018-03-20:
@Vavooon The link to the log you provided is broken. Please provide a functional link please.

Vavooon, 2018-03-20:
https://logs.px4.io/plot_app?log=f23bc221-f07d-403f-a4f8-61d80e7957aa
And here's raw file
https://drive.google.com/open?id=1H-7zfCWeuLMOSyWP0_bZQR7Rilzm5kDJ

closed 2019-02-09 by stale[bot]

fabrizioschiano, 2021-03-03:
Has anyone solved this problem? I am having the same issue

bresch, 2021-03-04:
> Has anyone solved this problem? I am having the same issue

Do you have this issue using master? If yes, please re-open a new issue.