PX4/PX4-Autopilot #9030: **Feature Request**: block arming after sensor calibration to require system reboot prior to flight

Antiheavy, 2018-03-06:
I've seen it recommended many times that the system should get a full power-cycle reboot after sensor re-calibration.  In out own experience, the step change or dynamic motion after sensor re-calibration (especially Compass Cal) often causes EKF or sensor warnings.  We've had flights where the EKF got unhappy after takeoff due to a re-calibration during preflight, ultimately causing failsafe behavior and sometimes vehicle crashes.

We've experimented a lot with soft reset of the EKF and the whole firmware, but that often cause various other errors too - maybe certain external sensor may remain powered during a soft reset or state machine things aren't resetting and that messes with things.  The most reliable method of cleanly re-initializing the EKF after calibration is a full system power cycle.

**Proposal**:  after completion of a sensor calibration send the message "sensor calibration updated, reboot required before arming", If arming is attempted, block the arming and send the same message again.  A system flag may be required for this.

FYI @dagar @priseborough

LorenzMeier, 2018-03-06:
That makes total sense.

priseborough, 2018-03-06:
I support this proposal. The step change in sensor offsets combined with the motion associated with calibration can be outside estimator tuning limits and and number of states/variables external to the estimator that would need to be handled correctly make an enforced reboot the logical choice.

dagar, 2018-04-21:
I would propose trying to directly address these issues rather than a full power cycle.

1. The state machine shouldn't allow arming with any of these errors. If this isn't already the case in master after https://github.com/PX4/Firmware/pull/9193 it needs to be fixed.

2. The system should be aware of calibration status either as a new calibration_status message or possibly as an entirely new arming state.

3. The EKF should reinitialize after it sees the transition out of ARMING_STATE_CALIBRATION.

4. If the EKF has any initialization requirements we should monitor, enforce, and report them. 
   - no rotation? "Vehicle must be still while EKF is initializing"
   - @priseborough anything else? 
   - I don't know that it matters anymore, but we'd also have everything in place to ignore that for ARMING_STATE_IN_AIR_RESTORE

1 & 2 could be done first and used to immediately @Antiheavy's proposal immediately while 3 & 4 can come later.

Antiheavy, 2018-04-21:
> and number of states/variables external to the estimator that would need to be handled correctly make an enforced reboot the logical choice.

@priseborough makes an important point here as well.

dagar, 2018-04-21:
> > and number of states/variables external to the estimator that would need to be handled correctly make an enforced reboot the logical choice.
> 
> @priseborough makes an important point here as well.

He's not wrong, I just think it's a sign of a deeper problem if we can't find a way to do this cleanly/safely.

Antiheavy, 2018-08-06:
Related issue in the QGC repo from another user:  https://github.com/mavlink/qgroundcontrol/issues/6768#issuecomment-410752280

dagar, 2018-08-06:
Perhaps we should finally do something about this. I propose we add a CALIBRATION state to the arming state transition table.

EKF2 already monitors the arming state to know when it's safe to save parameters for things like the learned mag bias. On transition out of CALIBRATION we can have EKF2 reinitialize itself, including asking the user to hold the vehicle still if necessary.

Thoughts?

Antiheavy, 2019-01-26:
This issue is still valid.  Multiple solutions have been proposed here for consideration if/when someone is able to work on it.

Antiheavy, 2019-06-25:
still valid.  more graceful re-initialization after re-calibration is needed for commercial and consumer grade products.

Antiheavy, 2019-09-23:
Still valid issue.

dagar, 2020-09-01:
A different solution - https://github.com/PX4/Firmware/pull/15655

dagar, 2021-01-09:
I think we can consider this one effectively resolved for a couple reasons.

1. The estimator(s) are now aware of sensor calibration changes (accel/gyro/mag) synchronized with the actual change applied.
2. We actually reset the estimated mag bias on change.
3. Sensor calibrations (accel/gyro/mag) no longer involve zeroing out the old calibration first. The calibration procedures are always performed directly on raw data, and on success the calibration is updated from old -> new.

If there's still concern I'm happy to entertain additional mechanisms for further enforced user behaviour (requiring the user to stop spinning before completing mag cal?), but I'd like to see data first.

closed 2021-01-09 by dagar

Antiheavy, 2021-01-09:
hooray!