PX4/PX4-Autopilot #9037: VTOL takeoff does transition into wrong direction

RomanBapst, 2018-03-07:
When using a VTOL takeoff mission item to combine takeoff and transition the vehicle does not consider turning to a heading which is equal to the bearing to the next waypoint. Instead it will transition into the current heading which I think is not expected by the user.

This issue does not occur when using a separate takeoff and transition command;.

We reproduced this two times using the Convergence tiltrotor (SYS_ID = 13012)

Here's the log
https://logs.px4.io/plot_app?log=be4b8af6-31b8-4c01-997a-d6f848ab7839

RomanBapst, 2018-03-07:
@dagar Could you please have a look at this? I assume the problem is here
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission.cpp#L612

dagar, 2018-03-07:
As discussed, I suspect this is intentional and we need to implement VTOL_TRANSITION_HEADING as defined in the Mavlink spec.

[image]

[image]

I'm extremely uncomfortable with the current complexity of the Mission handling code where this magic happens (https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission.cpp#L510-L582). If we're going to change anything here we need to cleanly refactor this into something comprehensible.

tubeme, 2018-03-09:
We can report the same behavior even outside mission mode. There is no heading (yaw) control during transition, so in strong weather if there is some cross wind during transition for 5-10 seconds, even in Position mode the VTOL could dangerously yaw off intended course and crash in a tree or building even on one occasion it flew directly toward the pilot. Good point made by @RomanBapst

closed 2019-02-09 by stale[bot]