PX4/PX4-Autopilot #9038: Intel Aero RTF: no baro0, no mag0

jdiament95, 2018-03-07:
I flashed the intel aero with PX4 1.7.3 (from my laptop) QGC is not able to detect the barometer. When I revert to PX4 1.6.5 (from the preloaded binary on the aero), QGC is able to successfully detect the barometer.

Anyone experiencing similar problems? Or have any suggested solutions?

LorenzMeier, 2018-03-07:
Thanks for the report! I've tagged the Intel lead for the platform and I'm sure she will investigate this quickly.

padmashree9110, 2018-03-08:
@jdiament95 I flashed 1.7.3 version and I don't see any issue. I checked device nodes as well inside nuttx shell, there is a baro node .

jdiament95, 2018-03-08:
@padmashree9110 can you report how you go about flashing the firmware? 

I'm flashing the firmware from my Mac laptop to the aero over the aero's access point wifi. First, I had to modify `Firmware/Tools/aero_upload.sh` with the proper macOS syntax for the local path names. Then I ran `make aerofc-v1_default upload` and successfully flashed the firmware to the aero. Finally, I confirm that I'm running the proper version because `aero-get-version.py` returns: Firmware: 1.7.3

padmashree9110, 2018-03-09:
If aero-get-version is showing 1.7.3 it will not be issue with flashing. Can you please share the screenshot of QGC with 1.7.3 and 1.6.5.

jdiament95, 2018-03-10:
for PX4 Firmware Version: 1.7.3dev, I get the error message "PREFLIGHT FAIL: NO BARO SENSOR 0." When I do "ls /dev" in a nuttx shell, I can confirm that there is NO baro0 listed. 

I don't have any of these problems for PX4 Firmware Version: 1.6.5dev

LorenzMeier, 2018-03-10:
@anitha-suresh Any updates on this?

sugnanprabhu, 2018-03-12:
@LorenzMeier as @padmashree9110  mentioned this is issue is not observed here

@jdiament95 when you say 1.7.3dev, is it release or are you using the development branch based on 1.7.3?
Can you please share the output of ver git in nsh shell

anitha-suresh, 2018-03-12:
Right, I see that there 'ls /dev in Nuttx shell' shows baro0 and also QGC doesn't show up any error. The PX4 FW version on which we verified is 1.7.3

jdiament95, 2018-03-12:
nsh> ver git
FW git-hash: 47e5382e7efa68bc3f34fd767a1a76193610112c
FW version: 1.7.3 0 (17236736)
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: b18053574bf41712cef93e31bf178518f451a350

padmashree9110, 2018-03-13:
@LorenzMeier @jdiament95 
I verified earlier with 1.7.3 release tag, issue is not seen.
With the commit id 47e5382, issue is seen. Issue is because of regression due to "[c20594c](https://github.com/PX4/Firmware/commit/c20594cd0ea23b44ed77bb6d96571b6277f4592a)"

It is resolved in master with this commit, [8d4036](https://github.com/PX4/Firmware/commit/8d4036df7d1899b1f883a4b7d78cf6e195b11a1d)

Please use either release tag 1.7.3 or tip of the repo.

closed 2018-03-13 by LorenzMeier