PX4/PX4-Autopilot #9039: Intel Aero RTF: using FC mag

jdiament95, 2018-03-07:
This issue (`https://github.com/intel-aero/meta-intel-aero/issues/221`) describes that the firmware uses the mag attached to the aero's GPS module. Is there any way to configure the firmware such that it uses the magnetometer attached to the aero flight controller as shown in the linked diagram: `https://github.com/intel-aero/meta-intel-aero/wiki/94-(References)-Sensors-GPIOs-LEDs-CAN-bus`?

LorenzMeier, 2018-03-07:
That would typically deteriorate the flight performance as there is more magnetic interference on the board - any particular reason you want this?

@anitha-suresh The Intel docs might need to be clarified.

jdiament95, 2018-03-07:
I snapped the GPS mast on my aero, so I wanted to use the internal mag instead of the one integrated on the GPS chip. Furthermore, I'm doing exclusively indoor flight using optical flow -- so I wanted to detach the GPS altogether.

But if the FC mad performance is compromised, do you think that mounting the GPS/mag chip on top of the compute board enclosure suffer from the same magnetic interference throughout flight?

zehortigoza, 2018-03-23:
You can dig into the PX4 git history and find the patch that remove the internal compass but be aware of what Lorenz said.
Mount direct on top of the compute board will not be optimal but is better than the inside one.

closed 2019-02-09 by stale[bot]