PX4/PX4-Autopilot #9040: Cannot see  	landing_target_estimator in QGroundControl

zquang, 2018-03-08:
HI all,

I am testing the Precision Land modules using IRLock sensor.
I was able to test in Gazebo simulation (make posix_sitl_default gazebo_iris_irlock). I can takeoff and auto:precland in the simulator. In QGC, it is possible to change parameter of "Precision Land" and  	"Landing target estimator"

I compiled the latest FW for Pixhawk 2 (Cube) from source (make px4fmu-v3_default) and upgrade FW in QGC. I can see parameters for "Precision Land" but cannot see "Landing target estimator".

As far as I know, "Landing target estimator" get information from IRLock sensor (Pixy camera) then output x-y offset of the landing target. It seem that this module is missing in the "real" FW.

Many thanks

mhkabir, 2018-03-08:
The estimator is included, but not started at boot. You will need to modify startup scripts or add an extras script on the sdcard.

closed 2018-03-08 by zquang

yl5006, 2018-03-09:
the latest QGroundControl can not get params in Gazebo simulation ,do you have this problem?

zquang, 2018-03-09:
I use QGC binayry installation from https://s3-us-west-2.amazonaws.com/qgroundcontrol/latest/QGroundControl.tar.bz2
It can get params from Gazebo, but not from real Pixhawk 2 (FW development version)

scheysej, 2023-06-15:
(For anyone having a similar issue)
I found that in QGC you have to go to the MAVLink console and type `landing_target_estimator start` and then you are able to see all of the irlock params. I was using a Pixhawk 2 (BlackCube).

In order to get this to start on boot...
- Take the microSD out of the cube
- Put it into your computer and open its location in the file explorer
- Create a folder called `etc`
- In that folder create `extras.txt`
- In that text file write `landing_target_estimator start` and save it
- Plug it back into the cube and it should start on boot!