PX4/PX4-Autopilot #9042: Makefile:147: recipe for target 'posix_sitl_default' failed

mymultiverse, 2018-03-08:
I am trying to run 

    make posix_sitl_default gazebo 
but I am getting the following error

[3/7] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   thread
--   signals
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
--   chrono
--   atomic
-- Boost version: 1.58.0
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   timer
--   chrono
--   date_time
--   atomic
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
-- catkin DISABLED
Gazebo version: 7.11
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   timer
--   chrono
--   date_time
--   atomic
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yograj/src/Firmware/build/posix_sitl_default/build_gazebo
[4/7] Performing build step for 'sitl_gazebo'
FAILED: cd /home/yograj/src/Firmware/build/posix_sitl_default/build_gazebo && /usr/bin/cmake --build . && /usr/bin/cmake -E touch /home/yograj/src/Firmware/build/posix_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-build
[1/52] Building CXX object CMakeFiles/mav_msgs.dir/Float.pb.cc.o
[2/52] Building CXX object CMakeFiles/mav_msgs.dir/opticalFlow.pb.cc.o
[3/52] Building CXX object CMakeFiles/mav_msgs.dir/CommandMotorSpeed.pb.cc.o
[4/52] Building CXX object CMakeFiles/mav_msgs.dir/lidar.pb.cc.o
[5/52] Building CXX object CMakeFiles/mav_msgs.dir/Int32.pb.cc.o
[6/52] Building CXX object CMakeFiles/mav_msgs.dir/SensorImu.pb.cc.o
FAILED: /usr/bin/c++   -Dmav_msgs_EXPORTS -isystem /usr/include/gazebo-7 -isystem /usr/include/sdformat-4.4 -isystem /usr/include/ignition/math2 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -I/home/yograj/src/Firmware/Tools/sitl_gazebo/include -I/home/yograj/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/include/gazebo-7/gazebo/msgs -I/usr/include/eigen3 -isystem /opt/ros/kinetic/include/opencv-3.3.1 -isystem /opt/ros/kinetic/include/opencv-3.3.1/opencv -I. -I/usr/include/eigen3/eigen3 -I/usr/local/include/OGRE -I/usr/local/include/OGRE/Paging -I/opt/ros/kinetic/include -std=c++11 -std=c++11 -Wno-deprecated-declarations -fPIC -MMD -MT CMakeFiles/mav_msgs.dir/Int32.pb.cc.o -MF CMakeFiles/mav_msgs.dir/Int32.pb.cc.o.d -o CMakeFiles/mav_msgs.dir/Int32.pb.cc.o -c Int32.pb.cc
In file included from Int32.pb.cc:5:0:
Int32.pb.h:12:2: error: #error This file was generated by a newer version of protoc which is
 #error This file was generated by a newer version of protoc which is
[... 59 lines cut ...]
  ^
Float.pb.h:14:2: error: #error your headers.
 #error your headers.
  ^
In file included from Float.pb.cc:5:0:
Float.pb.h:23:35: fatal error: google/protobuf/arena.h: No such file or directory
compilation terminated.
FAILED: /usr/bin/c++   -Dmav_msgs_EXPORTS -isystem /usr/include/gazebo-7 -isystem /usr/include/sdformat-4.4 -isystem /usr/include/ignition/math2 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -I/home/yograj/src/Firmware/Tools/sitl_gazebo/include -I/home/yograj/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/include/gazebo-7/gazebo/msgs -I/usr/include/eigen3 -isystem /opt/ros/kinetic/include/opencv-3.3.1 -isystem /opt/ros/kinetic/include/opencv-3.3.1/opencv -I. -I/usr/include/eigen3/eigen3 -I/usr/local/include/OGRE -I/usr/local/include/OGRE/Paging -I/opt/ros/kinetic/include -std=c++11 -std=c++11 -Wno-deprecated-declarations -fPIC -MMD -MT CMakeFiles/mav_msgs.dir/lidar.pb.cc.o -MF CMakeFiles/mav_msgs.dir/lidar.pb.cc.o.d -o CMakeFiles/mav_msgs.dir/lidar.pb.cc.o -c lidar.pb.cc
In file included from lidar.pb.cc:5:0:
lidar.pb.h:12:2: error: #error This file was generated by a newer version of protoc which is
 #error This file was generated by a newer version of protoc which is
  ^
lidar.pb.h:13:2: error: #error incompatible with your Protocol Buffer headers. Please update
 #error incompatible with your Protocol Buffer headers.  Please update
  ^
lidar.pb.h:14:2: error: #error your headers.
 #error your headers.
  ^
In file included from lidar.pb.cc:5:0:
lidar.pb.h:23:35: fatal error: google/protobuf/arena.h: No such file or directory
compilation terminated.
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
Makefile:147: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 1

dagar, 2018-03-09:
You seem to be missing dependencies for gazebo_sitl. Did you follow the instructions for installation from here? https://dev.px4.io/en/setup/dev_env.html

mymultiverse, 2018-03-09:
I am following [this](https://github.com/mymultiverse/ROSBasics#px4-installation). As I followed these steps earlier and I worked perfectly. 

I am able to run the simulation with 

    roslaunch px4 posix_sitl.launch
and 
    roslaunch px4 multi_uav_mavros_sitl.launch

mymultiverse, 2018-03-11:
I also reset the [development toolchain](https://dev.px4.io/en/setup/dev_env_linux.html#gazebo-with-ros) but again same error.

mymultiverse, 2018-03-14:
Update on this I can make using jmavsim successfully.  

       make posix jmavsim

mymultiverse, 2018-03-15:
This problem seems to be related to Gazebo and protobuf-compiler. But I installed the correct version of the dependency as [suggested](https://dev.px4.io/en/setup/dev_env_linux_ubuntu.html). 

    sudo apt-get install protobuf-compiler libeigen3-dev libopencv-dev -y

mymultiverse, 2018-04-03:
I followed development tool [chain](https://dev.px4.io/en/setup/dev_env_linux.html) setup for Gazebo with ROS on completely newly installed Ubuntu 16.04 .  Now it is working fine on new system. Not sure but there might be Gazebo version conflicts.

closed 2018-04-03 by mymultiverse