PX4/PX4-Autopilot #9059: Barometer compilation error- {solved by renaming variables external in bmp280_SPI constructor}

shivam5594, 2018-03-12:
[1/5] Building CXX object src/drivers/barometer/b...CMakeFiles/drivers__bmp280.dir/bmp280_spi.cpp.obj
FAILED: /usr/bin/arm-none-eabi-g++   -DBUILD_URI=localhost -DCONFIG_ARCH_BOARD_PX4FMU_V2 -DMODULE_NAME=\"bmp280\" -DPX4_MAIN=bmp280_app_main -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -isystem NuttX/nuttx/include/cxx -isystem NuttX/nuttx/include -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/apps/include -I. -Isrc -Isrc/modules -I../../src -I../../src/drivers/boards/px4fmu-v2 -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -Iexternal/Install/include -fno-common -ffunction-sections -fdata-sections -mcpu=cortex-m4 -mthumb -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fno-common -ffunction-sections -fdata-sections -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wno-overloaded-virtual -Wreorder -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Os -DNDEBUG   -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Wframe-larger-than=1024 -MMD -MT src/drivers/barometer/bmp280/CMakeFiles/drivers__bmp280.dir/bmp280_spi.cpp.obj -MF src/drivers/barometer/bmp280/CMakeFiles/drivers__bmp280.dir/bmp280_spi.cpp.obj.d -o src/drivers/barometer/bmp280/CMakeFiles/drivers__bmp280.dir/bmp280_spi.cpp.obj -c 

> **../../src/drivers/barometer/bmp280/bmp280_spi.cpp
> ../../src/drivers/barometer/bmp280/bmp280_spi.cpp: In constructor 'BMP280_SPI::BMP280_SPI(uint8_t, uint32_t, bool)':
> ../../src/drivers/barometer/bmp280/bmp280_spi.cpp:90:67: error: declaration of 'external' shadows a member of 'this' [-Werror=shadow]
>  BMP280_SPI::BMP280_SPI(uint8_t bus, uint32_t device, bool external) :
>                                                                    ^
> compilation terminated due to -Wfatal-errors.**

cc1plus: all warnings being treated as errors
ninja: build stopped: subcommand failed.
Makefile:153: recipe for target 'px4fmu-v3_default' failed
make: *** [px4fmu-v3_default] Error 1

bigbellmercy, 2018-03-20:
This error happens in fw v1.7.3.
But it did not happens in master (v1.7.4beta?).

dagar, 2018-03-21:
Fixed.

closed 2018-03-21 by dagar

IeiuniumLux, 2018-07-11:
I still see this error in v1.8.0-rc0 and v1.8.0

[ 55%] Building CXX object src/drivers/barometer/bmp280/CMakeFiles/drivers__bmp280.dir/bmp280_i2c.cpp.obj
/home/safe50/src/Firmware/src/drivers/barometer/bmp280/bmp280_i2c.cpp: In constructor 'BMP280_I2C::BMP280_I2C(uint8_t, uint8_t, bool)':
/home/safe50/src/Firmware/src/drivers/barometer/bmp280/bmp280_i2c.cpp:74:66: error: declaration of 'external' shadows a member of 'this' [-Werror=shadow]
 BMP280_I2C::BMP280_I2C(uint8_t bus, uint8_t device, bool external) :
                                                                  ^
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
src/drivers/barometer/bmp280/CMakeFiles/drivers__bmp280.dir/build.make:86: recipe for target 'src/drivers/barometer/bmp280/CMakeFiles/drivers__bmp280.dir/bmp280_i2c.cpp.obj' failed
make[3]: *** [src/drivers/barometer/bmp280/CMakeFiles/drivers__bmp280.dir/bmp280_i2c.cpp.obj] Error 1
CMakeFiles/Makefile2:2575: recipe for target 'src/drivers/barometer/bmp280/CMakeFiles/drivers__bmp280.dir/all' failed
make[2]: *** [src/drivers/barometer/bmp280/CMakeFiles/drivers__bmp280.dir/all] Error 2

bigbellmercy, 2018-07-13:
Al, for developers to know this problem your comment should be attached in
the post.

Your below comment was sent to only me by email.

cosmicog, 2018-12-19:
Hello guys, I think instead of following change-log to find and fix this issue, one should do [this](https://github.com/PX4/Firmware/commit/be52dcf5e36b651eda778076a5b2b9c71e33260c) change to the file `src/drivers/bmp280/bmp280_spi.cpp` under `v1.7.3` in order to compile successfully. 

I'm leaving this comment here to just help `v.1.7.3` users since the issue is closed for newer versions (and file path changed) and the fixing commit doesn't include a "`fixes #`" reference to this issue.