PX4/PX4-Autopilot #9060: EKF baro hgt timeout - reset to baro

Ahrovan, 2018-03-12:
**Bug Report**

 - flight log : [log](https://review.px4.io/plot_app?log=b3ce673f-cd61-4217-8866-f1b0350f74fb)
 - Crash after switch to Mission mode in Waypoint 4.
 - F450 quad copter frame + Ublox M8N + PX4 v1.7.3 Stable Release
 - Tested with SITL+Gazebo and without any problem.

[image]

[image]

[image]

xdwgood, 2018-03-14:
@Ahrovan  I just quickly analyzed the log.
Look this : 
[image]
Around 07:46, GPS data stopped updating.Before 07:46, the uav's Angle/position was tracked normally. 
So I think the GPS module is falling off in the air. Do you check the GPS module after the crash?　:smiley:

Ahrovan, 2018-03-17:
This problem occurred today with [link](https://review.px4.io/plot_app?log=2ac91f97-e3a1-4a10-8ea4-7e16918b4574) log file...

Ahrovan, 2018-03-17:
@xdwgood The QUAD COPTER suddenly loses its height (in this time I used Lidar lite v3 too)

xdwgood, 2018-03-19:
@Ahrovan  
yes,I found a problem with altitude control before landing(time:04:21).　see picture: 
[image]
It may be due to the pressure effect near the ground.　So it didn't crash, right?　@MaEtUgR  Is the speed and height error of the z direction before landing(time:04:21) acceptable?

@LorenzMeier  The customer used the most recent PX4 v1.7.3 Stable Release.　So I don't know if commander's low stack problem needs your attention.　I also have a suggestion: the ｄｒｏｎｅ is not allowed to enter `NAVIGATION_STATE_AUTO_MISSION` before take-off.　
[image]

Ahrovan, 2018-03-19:
@xdwgood Thanks . I tried to prevent crash with changing the flight mode to "Altitude Mode": for now, there is no solution to the problem, I agree with you for altitude problem. for today I tested again with this result:
[review log](https://review.px4.io/plot_app?log=2cd29222-5da3-4485-a87f-f044564a1933)
- checked possible battery problems (_**ok, no problem**_)
- checked possible range finder problems (_**ok, no problem**_)
- checked possible waypoint's problems (_**ok, no problem**_)
- calibrated all sensor again.

Altitude estimator show "GPS Altitude" and "distance sensor" descend in 2:00 - 2:20:
[image]
[image]
 but "Fuse Altitude Estimator" and "barometer Altitude" show ascend Altitude?!
[image]

Ahrovan, 2018-03-19:
@MaEtUgR Thank you. I will review and report on the cases today.
- I used "Mount Anti-vibration Plate Shock"
- I used generic "Pixhawk PX4 Autopilot PIX 2.4.6 32Bits" board. inside it, is small foam 
- My barometer is MEAS MS5611

Ahrovan, 2018-03-30:
New flight with new Hex Frame (All Hardware changed, Just Pixhawk stay):
https://review.px4.io/plot_app?log=33bb60a6-ce79-47ed-8573-a101b60fae44

crash due:
EKF baro hgt timeout - reset to baro

Ahrovan, 2018-03-30:
see [ecl](https://github.com/PX4/ecl/blob/ba2b9dfdd96d50d697165407b88b5bc94cdef84c/EKF/control.cpp#L590)

	// check if height is continuously failing becasue of accel errors
	bool continuous_bad_accel_hgt = ((_time_last_imu - _time_good_vert_accel) > (unsigned)_params.bad_acc_reset_delay_us);

	// check if height has been inertial deadreckoning for too long
	bool hgt_fusion_timeout = ((_time_last_imu - _time_last_hgt_fuse) > (uint64_t)5e6);

	// reset the vertical position and velocity states
	if (hgt_fusion_timeout || continuous_bad_accel_hgt) {
...
		// handle the case where we are using baro for height
		if (_control_status.flags.baro_hgt) {
...
			if (reset_to_gps) {
...
				ECL_WARN("EKF baro hgt timeout - reset to GPS");

			} else if (reset_to_baro) {
...
				ECL_WARN("EKF baro hgt timeout - reset to baro");

			} else {
...
			}
		}

priseborough, 2018-04-03:
@Ahrovan The is no EKF issue until after the vibration levels suddenly become extreme. Following plot shows IMU accel scaled to units of g and the vertical velocity and position innovations:

[image]

You need to fix your hardware.

closed 2018-04-03 by priseborough

Ahrovan, 2018-05-03:
@priseborough It looks like you checked the acceleration after the crash?

priseborough, 2018-05-03:
Then that means there was no EKF issue during flight.

Ahrovan, 2018-05-03:
A sudden change occurs between the height of the baro and the gps, causing a sudden loss of altitude in the situation that has been raised.

priseborough, 2018-05-03:
The baro output is inconsistent with other sensors. It looks like a faulty baro sensor was responsible.

Ahrovan, 2018-05-03:
This problem occurred on several hardware, it seems to me incorrect. Because there is no problem in manual mode or altitude. Except in navigation mode with gps.

priseborough, 2018-05-03:
What autopilot hardware are you using?

Ahrovan, 2018-05-03:
@priseborough Pixhawk 1 Flight Controller
<img src="https://docs.px4.io/assets/hardware/hardware-pixhawk.png
" alt="Pixhawk" height="42" width="42">

tkris, 2018-06-25:
@Ahrovan is your Problem solved ?

I had yesterday the same issue 2 times with my S500 Quadcopter (Pixracer Controller) - EKF baro hgt timeout - reset to baro. My Quad suddenly gained altitude in altitude or position hold mode... very shocking experience. I am running PX4 1.8.0 but i had the same Problem with Arducopter.

Everytime the S500 Quad would arm and fly nice in stabilized mode but when i switch to altitude or position hold the problem occurs.

Flight logs:
https://logs.px4.io/plot_app?log=7e17a3e1-b15f-47fc-93fe-29bde57b09db
https://logs.px4.io/plot_app?log=d98bfd03-f428-416b-9130-15a1bcc3a01b

Maybe someone can help me out. I don´t know if my pixracer is bad or my GPS Module. Today i want to try a soft mount of the FC.

priseborough, 2018-06-25:
@tkris This  is intended for the reporting of SW defects, not resolving problems with hardware, installation and setup.

Ahrovan, 2018-06-26:
@tkris @priseborough The change in the hardware of the FC does not have an effect on the problem and this problem occurs again(So that we tested with versions Pixhawk 1 lite and Pixhawk 1). 
The latest change we have made in the work plan to solve this problem is to change the gps module and to strengthen the power connections of motors. Changing the FC damper did not affect the problem. We look forward to working with you to solve this problem.