PX4/PX4-Autopilot #9061: Pixhawk/PX4FMUv2 startup without external mag fails

philipoe, 2018-03-12:
**Bug Report**

#### Issue: 
On Pixhawk with rev. 3 microprocessor (i.e. compiled via px4fmu-v3_default), current master firmware (https://github.com/PX4/Firmware/commit/63121b74dfe71ea63ad6863b42e39c6fb85ca40b) and stock startup script (rcS with airframe "standard plane" / Autostart ID 2100), bootup without an external magnetometer does not work because lsm303d fails. The system is a 64bit Ubuntu 16.10LTS. The exact log output for the configuration is:

> HW arch: PX4FMU_V2
> HW type: V2
> HW version: 0x0009000B
> HW revision: 0x00000000
> FW git-hash: 63121b74dfe71ea63ad6863b42e39c6fb85ca40b
> FW version: 1.7.3 0 (17236736)
> OS: NuttX
> OS version: Release 7.22.0 (118882559)
> OS git-hash: efc3d9e92bd10623ab3025f012fd7d481e1a7267
> Build datetime: Mar 12 2018 13:40:08
> Build uri: localhost
> Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
> MFGUID: 393839343335511000380045
> MCU: STM32F42x, rev. 3
> UID: 380045:33355110:39383934

#### Steps to reproduce the problem:
 - By default, l3gd20 will start but lsm303d will not start because it is not part of the px4fmu-v3_default config file.  The console output is as simple as "lsm303d: Command not found".
 - When just adding drivers/imu/lsm303d to cmake/configs/nuttx_px4fmu-v3_default.cmake, suddenly neither l3gd20 nor lsm303d start up anymore after a power cycle. Console output is:

> INFO  [mpu6000] accel cutoff set to 30.00 Hz
> INFO  [mpu6000] gyro cutoff set to 30.00 Hz
> WARN  [mpu6000] no device on bus #5 (SPI4)
> WARN  [mpu9250] probe failed! 104
> WARN  [mpu9250] no device on bus 3
> WARN  [mpu9250] probe failed! 255
> WARN  [mpu9250] no device on bus 4
> ERROR [l3gd20] driver start failed
> WARN  [lsm303d] SPI init failed

However, when doing a soft reboot (via nsh `reboot` command), both sensors start up correctly!

#### Questions:
I know that it is _preferred_ to use an external mag, but why is lsm303d completely disabled in nuttx_px4fmu-v3_default.cmake ? And why does it not start after a power cycle although it does start after nsh's `reboot?`

#### Notes:
This was tested on 3 different Pixhawks, all with revision 3 microprocessors and thus 2MB flash (on the PCB, these are just marked as "PX4FMU 2.4"). Same issue on all of them. However, on a fourth Pixhawk (which also has a rev. 3 microprocessor) this works every time (PCB mark is "PX4FMU 2.4.6", could be an mRo Pixhawk)! Is this a known problem of some hardware versions of Pixhawk?

@dagar @ASM3

tops4u, 2018-03-12:
Check current Master (IMHO the above is not current Master), since there was a Fix for SPI Bus Reset, which fixes the LSM303D not found after a Soft Reboot (Warm Start): https://github.com/PX4/Firmware/issues/8992

philipoe, 2018-03-12:
The "above" (https://github.com/PX4/Firmware/commit/63121b74dfe71ea63ad6863b42e39c6fb85ca40b) is the current master as of right now, and the issue unfortunately persists.

LorenzMeier, 2018-03-14:
@philipoe Another fix went into master from @dagar - could you re-test?

philipoe, 2018-03-14:
Tested on 2 different Pixhawk units during ~10 hard-reboots, worked every time. 

Thanks @dagar for the quick fix!

closed 2018-03-14 by philipoe