PX4/PX4-Autopilot #9066: QGC parameter cache hit failing due to parameter intermittently missin

DonLakeFlyer, 2018-03-12:
You need to use latest QGC daily to test against. Not windows version since windows drops are f'ed up right now.

* Run latest master ```make posix_sitl_default jmavsim -j16```
* Boot QGC, Exit QGC. Do this three time to correctly primes the parameter cache.
* Exit firmware SITL
* Boot QGC, turn on ParameterManagerDebugCacheFailure logging
* Start firmware SITL

You will get this log output:
```
ParameterManagerLog: Attemping load from cache
ParameterManagerLog: Parameters cache match failed /Users/Don/.config/org.qgroundcontrol/ParamCache/1_1.v2
Parameter in cache but not on vehicle componentId:Name 1 "CAL_ACC_PRIME"
Parameter in cache but not on vehicle componentId:Name 1 "CAL_BARO_PRIME"
Parameter in cache but not on vehicle componentId:Name 1 "CAL_GYRO_PRIME"
Parameter in cache but not on vehicle componentId:Name 1 "CAL_MAG_PRIME"
Parameter in cache but not on vehicle componentId:Name 1 "COM_ARM_MAG"
```

Which indicates the parameter set is suddenly differernt from what is was before. If you do things in various order of start QGC before after SITL or leave sitl running you can also hit a case where that same of parameter come back from the vehicle, but they are not in the qgc cache.

All of this leads to intermittent parameter cache hit failure depending on the order in which you do things.

dagar, 2018-03-14:
These parameters are set by the calibration code, and then used in the preflight checks (in commander). Some of them aren't even used. I'm guessing this is a race condition between the first iteration of the preflight checks and the QGC param sync.

This makes me wonder if the param hash isn't working correctly. It sounds like it isn't factoring in which params are actually active.

DonLakeFlyer, 2018-03-14:
> I'm guessing this is a race condition between the first iteration of the preflight checks and the QGC param sync.

That would make sense as to why things change around depending on when QGC connects. Before or after preflight completes. If you start QGC and then start vehicle, the parameters returned to QGC in initial query do not contain these params. Whereas if you connect QGC to a vehicle already booted and past startup, then the initial query by QGC does contain these parameters. Sounds like they may need to be marked non-optional on firmware side so they are not race condition dependant?

> It sounds like it isn't factoring in which params are actually active.

I don't know what the means. The QGC hash is based off of the params received in the intial param_request_list, minus the volatile ones. If that list differs from connect to connect the hash simply won't work correctly.

dagar, 2018-03-14:
I'm talking about the hash firmware side. There's nothing stopping someone from starting an additional module much later that marks a bunch of params active. Do we have anyway of broadcasting that to the ground?

We can go out of our way to make sure all the important parameters are marked active at boot before QGC ever connects, but there's always the potential for more issues if this isn't factored in.

DonLakeFlyer, 2018-03-14:
Ah, I see. Tricky. Vehicle would have to send parameters to ground that show up on the fly after the first param_request_list. Then they get into the ground hash. But then that breaks the ability to debug CRC failures on the ground side.

closed 2019-02-09 by stale[bot]