PX4/PX4-Autopilot #9072: **Bug Report** : Mode change errors

JonReacher, 2018-03-13:
Reproducible:  PX4 1.7.3   QGC 3.3.0

-Set flight modes in QGC
-Switch between flight modes using R/C Radio
-Change at least one of the flight modes
-When switching between flight modes using R/C Radio, sometimes the old flight mode will be 
  entered instead of the new.

This caused a crash/ total loss when switching from mission to stabilized mode during landing. The log shows a third mode was entered briefly. I have only Mission and Stabilize modes on a 2-way switch, no other mode should have been possible. I have since reproduced at will.

I'm assuming this is a flight controller firmware issue, not a GCS issue?

Log: https://logs.px4.io/plot_app?log=061bf8ae-0a06-43d9-9f0e-5be63e1c6e49

proficnc, 2018-03-13:
What exact flight modes did you have set in QGC?
And how is your radio set up?
What was the third flight mode that was entered?

JonReacher, 2018-03-13:
I had Mission and Stabilize set in QGC flight mode setup. It was mapped to a 2-way toggle switch on my Taranis radio. I'm not sure what flight mode it is for sure. 

The mode is stated on the log for the Mission and Stabilize, but not for the third. All of the mode colors look lighter than the legend, to me anyway. So maybe Manual mode?

JonReacher, 2018-03-20:
@proficnc The 6th flight mode slot is what I changed into, which is set to Stabilized. The parameters list the range for this slot as 1800-2000 PWM. My Taranis goes to up to 2012, and is at 2012 when I flip the switch. Do you think it's possible that the 2012 PWM signal value exceeded the range for the 6th slot and is acting up?

closed 2019-02-09 by stale[bot]