PX4/PX4-Autopilot #9089: changing from offboard to the manual control mode

raywang-nx, 2018-03-16:
**Feature Request**

https://logs.px4.io/plot_app?log=23636d5d-dd1f-4f47-9ea0-2ca4a840cf39
https://logs.px4.io/plot_app?log=978d6037-b9ec-4a5d-bfd9-59cedf40adaf
https://logs.px4.io/plot_app?log=0d72d966-7bdc-41b4-9a4d-1e02bf019047
https://logs.px4.io/plot_app?log=9b5b6acb-362a-44de-ac63-b0ef7638df8b
https://logs.px4.io/plot_app?log=a770d83b-0326-44e1-a55f-8f87f5932943
https://logs.px4.io/plot_app?log=ed8c4189-7417-4cbd-b2a1-40a2298f00af

here are the six log files of the pixhawk when i met the problem, and i don't know which is the one i need, so i put them all there.

i got a task here, not quite difficult, but a little bit complex.
i got a pixhawk and joule running mavros. the job of joule is to control the the quadrotor to move forward, and avoid the obstacles, which means if there are obstacles, the joule will send the instructions to make it fly up, down, left and right. 
and now my testing environment is the parking lot on the b3 floor. so the height is limited to about 2.5m. considering it will fly upward to about 2m or so to avoid the obstacle, so i just want it to fly as low as possiable so i change the safety_area. the steps i used is that firstly, use the controller to make the drone just fly off the ground, and then change to offboard mode to let the mavros and joule take control, it suppose to fly to about 0.2 metere above the ground.  in order to control it to land safely, after the drone fly higher, i change the offboard mode to the land mode to land safe and sound. but it rejected to the land mode, and now i got a broken drone and will repair it latter.... before that, i will change the offboard to the stabilized mode, but in that mode i didn't give enough throttle, so it fall down immediately and ..... you know what will happen... so i wanna ask that what can i do to make it land safely from the offboard mode?

while i am using the latest firmware v1.7.3, and ubuntu 16.04, the hardware is pixhawk and joule with telem2.

lamping7, 2018-03-16:
Why not stay in offboard and land with local position setpoints (i.e. with the z component)?

raywang-nx, 2018-03-16:
@lamping7 cause when in offboard, it contains a series of actions, including obstacles avoidance, guidance etc. but here i just want a record and some pictures of the obstacles avoidance, so there is no need to wait. and due to the b3, there will be much more dangerous if i let it fly longer.

closed 2019-02-09 by stale[bot]