PX4/PX4-Autopilot #9091: Unable to build v1.6.5 for Pixhawk

myeabhishek, 2018-03-16:
Hi,

I have checked out v1.6.5 for my drone as I was using it earlier as well. I am unable to build the same for my px4fmu-v2_default target. I get the following error. Kindly help!!

[3/503] Building NuttX for px4io-v2 with nsh
FAILED: cd /home/abhishek/all_ws/mFlightFirmware/build_px4fmu-v2_default/px4io-v2 && /bin/rm -rf /home/abhishek/all_ws/mFlightFirmware/build_px4fmu-v2_default/px4io-v2/NuttX/nuttx/nuttx-export && /usr/bin/make --quiet --no-print-directory -C /home/abhishek/all_ws/mFlightFirmware/build_px4fmu-v2_default/px4io-v2/NuttX/nuttx -r CONFIG_ARCH_BOARD=px4io-v2 export >nuttx_build.log && cmake -E touch /home/abhishek/all_ws/mFlightFirmware/build_px4fmu-v2_default/px4io-v2/nuttx_export.stamp
stdio/lib_sscanf.c: In function 'vsscanf':
stdio/lib_sscanf.c:419:28: error: this statement may fall through [-Werror=implicit-fallthrough=]
                       sign = true;
                            ^
stdio/lib_sscanf.c:420:21: note: here
                     case 'u':
                     ^~~~
cc1: all warnings being treated as errors
make[2]: *** [bin/lib_sscanf.o] Error 1
make[1]: *** [libc/libc.a] Error 2
[3/503] Building NuttX for px4fmu-v2 with nsh
FAILED: cd /home/abhishek/all_ws/mFlightFirmware/build_px4fmu-v2_default/px4fmu-v2 && /bin/rm -rf /home/abhishek/all_ws/mFlightFirmware/build_px4fmu-v2_default/px4fmu-v2/NuttX/nuttx/nuttx-export && /usr/bin/make --quiet --no-print-directory -C /home/abhishek/all_ws/mFlightFirmware/build_px4fmu-v2_default/px4fmu-v2/NuttX/nuttx -r CONFIG_ARCH_BOARD=px4fmu-v2 export >nuttx_build.log && cmake -E touch /home/abhishek/all_ws/mFlightFirmware/build_px4fmu-v2_default/px4fmu-v2/nuttx_export.stamp
stdio/lib_sscanf.c: In function 'vsscanf':
stdio/lib_sscanf.c:419:28: error: this statement may fall through [-Werror=implicit-fallthrough=]
                       sign = true;
                            ^
stdio/lib_sscanf.c:420:21: note: here
                     case 'u':
                     ^~~~
cc1: all warnings being treated as errors
make[2]: *** [bin/lib_sscanf.o] Error 1
make[1]: *** [libc/libc.a] Error 2
ninja: build stopped: subcommand failed.
Makefile:164: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 1

hamishwillee, 2018-03-16:
Are you using the supported version of GCC? https://dev.px4.io/en/setup/dev_env_linux_ubuntu.html#nuttx-based-hardware

myeabhishek, 2018-03-16:
I will check and let you know. I have installed the toolchain from px4 developer guide convenience scripts only.

myeabhishek, 2018-03-16:
Following is my version for GCC. Do I need to use an older version for 1.6.5???

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

myeabhishek, 2018-03-19:
@hamishwillee Do you have the documentation to install older versions of GCC with which I could successfully build 1.6.5?

hamishwillee, 2018-03-19:
@myeabhishek The way I would work this out is to review the docker file history: https://github.com/PX4/containers/commits/master/docker/px4-dev/Dockerfile_nuttx

myeabhishek, 2018-03-20:
Issue resolved. Thanks.

closed 2018-03-20 by myeabhishek

Thieso, 2018-04-13:
Can you tell how you resolved this since i have the same problem