PX4/PX4-Autopilot #9093: ROMFS not rebuilt correctly when ROMFS files are 'touch'ed

ksschwabe, 2018-03-16:
@dagar: related to discovery in PR: #9067.

If one of the files in ROMFS/px4_common/ (or what ever directory you specify as your src ROMFS directory) is touched, then the ROMFS is not rebuilt - it is rebuilt ONLY if an airframe is touched (or file that starts with a number). This is because the add_custom_command in "ROMFS/CMakeLists.txt" does not depend on the contents of the ROMFS/px4_common/ directory. It only depends on the air frames (files that start with a number).
```
file(GLOB_RECURSE init_airframes ${PX4_SOURCE_DIR}/ROMFS/${config_romfs_root}/*/[1-9]*)
add_custom_command(OUTPUT ${romfs_temp_dir}/init.d/rcS ${romfs_temp_dir}/init.d/rc.autostart
	COMMAND cmake -E copy_directory ${romfs_src_dir} ${romfs_temp_dir}
	COMMAND ${PYTHON_EXECUTABLE} ${PX4_SOURCE_DIR}/Tools/px_process_airframes.py
		-a ${romfs_temp_dir}/init.d
		-s ${romfs_temp_dir}/init.d/rc.autostart
		--board ${BOARD}
	COMMAND ${PYTHON_EXECUTABLE} ${PX4_SOURCE_DIR}/Tools/px_romfs_pruner.py
		--folder ${romfs_temp_dir} --board ${BOARD}
	DEPENDS
		${romfs_src_files}
		${init_airframes}
		${PX4_SOURCE_DIR}/ROMFS/${config_romfs_root}/init.d/rcS
		${PX4_SOURCE_DIR}/Tools/px_process_airframes.py
	)
```
**Note:** although, the add_custom_command() depends on `${romfs_src_files}`, this variable is never set anywhere; it is an empty dependency. I suspect that `${romfs_src_files}` was supposed be set with a `file(GLOB_RECURSE ...)`, but this has just not been done.

**Note:** As it currently stands, if you make a change to any of the `rc.*` files other than `rcS`, then the ROMFS will not be rebuilt.

ksschwabe, 2018-03-16:
What we could do is remove the `file(GLOB_RECURSE init_airframes ${PX4_SOURCE_DIR}/ROMFS/${config_romfs_root}/*/[1-9]*)` line and replace it with `file(GLOB_RECURSE romfs_src_files ${romfs_src_dir} ${romfs_src_dir}/*)`

We could then remove the dependence on `${init_airframes}`, since it would be redundant. And as far as I can see, its only purpose is to cause a rebuild of the ROMFS if an airframe file is touched.

Tell me if you think I should submit a PR for this.

dagar, 2018-03-16:
I think we might need to explicitly list all the files. Globbing generally doesn't work very well in cmake. It will miss added files (you need to know how to manually kick off another cmake configure), and it's perhaps even more annoying when you remove a file that was previously globbed.

dagar, 2018-03-18:
I'd be up for requiring all the ROMFS files to be explicitly listed in a CMakeLists.txt. With some simple directory hierarchy in place in won't be too overwhelming.

bkueng, 2018-03-19:
> I'd be up for requiring all the ROMFS files to be explicitly listed in a CMakeLists.txt

Agreed. cmake recommends not to use GLOB and it will improve incremental builds.

taileron, 2018-03-19:
current master after flashing into fmu_V4 (vtol) all tuning values get lost or don´t load, probably the same source as mentioned above. Now I flashed back to master from 8 th of march that still was o.k.

ksschwabe, 2018-03-19:
Conversation continued here:  #9107.

PR #9107 (2018-03-26): Lists ROMFS files explicity instead of using GLOB_RECURSE
Fixes #9093.

Previously, when ROMFS files that were not airframes were touched, the ROMFS
would not be rebuilt. The ROMFS files are now specified explicityl in a
CMakeLists.txt file that is located in the root ROMFS directory.

Now when one of the ROMFS files is touched the whole ROMFS is rebuilt.

When new files are added to the ROMFS, they need to be explicity added to
the CMakeLists in the ROMFS root directory.
closed 2018-03-26 by dagar