PX4/PX4-Autopilot #9108: PX4 1.7.3: Total crash in loiter.

tubeme, 2018-03-19:
Hello,

We've had a total crash today with 1.7.3 when we engaged loiter in FW flight. We could not find the reason behind this crash in the Log...

While we were flying 1.7.3 for a week we've had no unexpected problems until today.

VTOL deltaquad
PX4 1.7.3
QGC 3.3 Windows
AUAV X2.1 HW
Without airspeed sensor

When the pilot activated Hold the plane just nose dived without any control whatsoever.
During the nose dive the pilot switched to MC flight but did not switch back the loiter switch, so it did not work as expected to safe the VTOL. (We did not test this but what is happening when we switch the transition switch while in loiter?)
There are no HW problems with the machine.

LOG: https://logs.px4.io/plot_app?log=89bd8fa6-6506-4e9a-87d5-81ee2cc9243a

The log seems to be cut at the end... but the video shows it all.

Video: https://www.youtube.com/watch?v=7iCPhtmdQq8&feature=youtu.be

Any ideas?

dagar, 2018-03-19:
What's the history of this vehicle? Does it have previous successful flights manual/stabilized/loiter/mission/etc?

dagar, 2018-03-19:
[image]

The pink line is the manual control pitch input. Why is it close to max during manual flight? It seems like you have an enormous TRIM problem. In manual you were flying with about 80% throttle, but when starting an AUTO mode TECS initializes at FW_THR_CRUISE (50%), which seems way too low for this vehicle.

tubeme, 2018-03-19:
NO the pitch is because we are climbing to a safe altitude in the beginning of the flight, so we push manually the throttle to about 80% for the climb. Once we reach the desired altitude we engaged the loiter.

The motor is with a very very good ratio for the plane, even in this case the plane does not carry camera so it is even lighter.

In this configuration the plane can fly perfectly well with a FW_THR_CRUISE (42%) tested and guaranteed. We usually fly it in auto modes with 60% but this time we used 50%. With 50% throttle the plane is flying just perfect, we have a headroom of 10% throttle until we reach the problematic 40% throttle. At 40% throttle we begin to see swings (too low speed).

The CG of the plane is perfect as well, because we test it in each flying session prior the main test flights.

Is there possibility to have a relation with the TECS problems we've experiencing? https://github.com/PX4/Firmware/issues/9078

tubeme, 2018-03-19:
The plane did not have a single crash for two months with 1.6.5 and one week with 1.7.3 until today.

All modes tested to their extremes! we've had about 45 flight sessions with average about 4-6 flights each session. We've had also about 30 survey missions with very expensive camera. Thanks god we've had the camera removed for this flight...

dagar, 2018-03-19:
Can you post a good log for comparison?

RomanBapst, 2018-03-19:
Things checked so far
- [ ] nothing visible in system power
- seems like pitch control was starting to degrade shortly  before switch to loiter
- [ ] cpu and RAM usage seem ok
- log stops 1.5 seconds after switching into loiter when vehicle was still in the air

RomanBapst, 2018-03-19:
@tubeme I was watching the video and I think you can clearly see how the elevon close to the camera goes into the up position due to the pitch up command from the controller.
However, I can't really see if the servo on the other side is really doing anything.

RomanBapst, 2018-03-19:
Also I think you can see how the rate controller was still doing something during the dive because the elevon close to the camera starts to oscillate which is normals since you don't have an airspeed sensor and thus no scaling takes place.

tubeme, 2018-03-19:
Good flight.

https://review.px4.io/plot_app?log=c3b1a88d-227a-4a18-aa17-067c680591ec

dagar, 2018-03-19:
I'd say it's not moving. There's a roll rate everytime the pitch rate controller commands a change.

[image]

Roll is trying to compensate the entire time.

[image]

tubeme, 2018-03-19:
@RomanBapst I thought the same thing, if the other elevon stop working. But then the pattern in 

[image]

Should look different if servo malfunction. They should clip. Especially the broken servo channel. But here servos are just ok...

RomanBapst, 2018-03-19:
Well, we the other huge question is why did the logging all of a sudden stop? Doesn't sound like a coincident does it?

RomanBapst, 2018-03-19:
@tubeme I assume it can't be something stupid like the airflow pushing in- and out the SD card? It seems like you have something mounted on top of the wing.

RomanBapst, 2018-03-19:
@tubeme Also keep in mind that you don't see any clipping because the log stops too early. At the time the log stops the pitch error is not very large but clearly in the video you can see it decreasing until it's nose down.
For the vehicle to pitch down like that  we can come up with several failure options:
- one servo stopped working, the other one was pulling the elevon up pretty far as seen on the video
- something came loose in the plane, e.g. battery moving forward and the elevons did not have enough authority
- the log stops all of a sudden which could hint to a autopilot failure

tubeme, 2018-03-19:
@RomanBapst That is the GPS stand because of the MAG we had to move it away from electronics.

Agree this stopping of the logging is strange... But if it is related to the SD card this is strange too...

Also I thought that this might be the problem we are not seeing. But why the log stopped on the first place... 

Also from the video it is visible that despite the logging stopped the pilot switched back to MC mode and tried to recover the quad just until the plane crashed. So the autopilot somehow worked at that time, but not logging for some reason and this led to some other failure??

The batteries are secured well.

dagar, 2018-03-19:
Verifying the wiring would be another valuable clue. If the right elevon is output 4 (as stated in the 13006 airframe configuration), then it's the left elevon (output 5 - orange) that's supposedly moving wildly according to the log.

[image]

RomanBapst, 2018-03-19:
@tubeme I assume the SD card was still in the autopilot after the crash?

tubeme, 2018-03-19:
@RomanBapst It was!

dagar, 2018-03-19:
14:40:56
[image]

14:41:01
[image]

14:41:08
[image]

14:41:14
[image]

14:41:23
[image]

14:41:39
[image]

14:41:57
[image]

dagar, 2018-03-19:
The loss of a few seconds of logging is concerning, but it's something I've seen before. The pixhawk logger buffer is quite small (usually around 16 kB depending on configuration), and the default logging rate is probably ~ 40 kB/s. 

So I'm wondering if it's corruption of the fat filesystem.

dagar, 2018-03-19:
Have you powered up the pixhawk since? If not, it's still worth a try with an sdcard in and serial console attached. If the autopilot actually locked up or crashed the hard fault logger should have caught something. If the timing was right (aka wrong) it might be possible that the hardfault wasn't written to the sdcard before impact.

RomanBapst, 2018-03-19:
@tubeme @dagar Something you can clearly see is that in the log from the crash you have a constant offset between roll_rate_setpoint and roll_rate (I'm talking bodyrate).
Btw, not that it's relevant for you crash but your rate I gains are quite small, I think you can increase those in the future.

dagar, 2018-03-19:
@tubeme any chance you have the corresponding mavlink log from QGroundControl?

tubeme, 2018-03-19:
@dagar We powered it up. Even made some tests on the camera trigger.

How can I get this hard fault log?

dagar, 2018-03-19:
> How can I get this hard fault log?
Check for any additional log files on the sdcard.

tubeme, 2018-03-19:
@dagar @RomanBapst We just tested the servos with bottles of water on top. They push very hard and are in perfect condition. Our way of wiring on our controller is such that there is no possibility to have a faulty connection.

For worse we don't have the other logs by mistake and we did not save the mavlink log......  :( stupid mistake

tubeme, 2018-03-19:
I don't know if it related but there is unexpected behavior in 1.7.3 in Stabilized flight.

In PX4 1.7.3 we make the following experiment. We climb the plane to 100-200 meters. Then we release the throttle to 0 and want to glide. Then the PX4 removes the controls from the Pitch and Roll stick and we cannot control the plane, and at the same time the nose starts to pitch down, slowly for about 3-4 seconds the pitch down does not stop and the plane starts to nose dive with 60 deg angle and without pitch and roll control. There are two option in this situation. Either push the throttle (pretty dangerous given the plane has a lot of speed in this nose dive) or transition in quad to save the plane. We used the quad at least 4 times for a week to save the plane.

In 1.6.5 it was in reverse. When we release the throttle to 0 the plane started to pitch up, sometimes at dangerous angels almost at the brink of stall. But at least we had control and could push the nose down in order to gain speed. If we leave it in 1.6.5 without control in the glide it will most likely will stall.

I cannot grasp yet what is making this dramatic change of behavior between the two versions of PX4...

If it not related sorry. I will delete it. or make another issue.

dagar, 2018-03-19:
Yes please create another issue if you'd like to pursue it. Showing side by side logs with that type of corresponding explanation can be quite helpful. 

For this issue one crude thing you could do on the bench with everything set up the same (camera same location and recording) is manually move the control surfaces and call out the corresponding PWM values. Some simple calibration mapping which motor back to which servo output, and rough PWM ranges would put the video in context. Watching it yet again I still find it hard to believe the left elevon is moving at all.

tubeme, 2018-03-19:
I watched the video at least 20 times and the servo is moving.

Here is another video that is with better light condition and you can better see the oposit elevon moving. Then again watch the crash video and you will se that the servo is moving. The camera is wide angle.

https://www.youtube.com/watch?v=HpgNfv1rG44&feature=youtu.be

dagar, 2018-03-19:
Actually in that video I can definitely see it moving, and it's easier to see in motion than these still frames. In the crash video the left elevon servo still looks rock solid to me.

17:09:42
[image]

17:09:53
[image]

17:10:23
[image]

17:10:24
[image]

Can you verify the motor connections at the time of the crash? 
https://logs.px4.io/plot_app?log=89bd8fa6-6506-4e9a-87d5-81ee2cc9243a

Was output 4 the left or right elevon servo?
Was output 5 the left or right elevon servo?

Antiheavy, 2018-03-20:
integrator windup?

tubeme, 2018-03-21:
I can confirm that 4 (ch5) is the elevon close to the camera.
and 5 (ch6) is the elevon far from the camera.

ie. it is like in the original configuration so no reverse was used.

tubeme, 2018-03-21:
@dagar You can better see the elevon's movement at the moment of transition to both flights, the crash one and the previous one. 

What do you mean by motor connections? If you mean the pusher, it just flew out and was about 10m away from the plane.

tubeme, 2018-03-21:
Here is the original video:

https://drive.google.com/file/d/1JfpMQiwmufwSKTfRoeub6MdiUOEhdklv/view?usp=sharing

I Played it with VLC and slowed it down to 35%. You can clearly see the eleveon moving. I can see that there is a critical moment regarding the pitch. Once we climbed out and enabled loiter I can clearly see that the elevons are making right turn movement, the close elevon is deflecting and the opposite one is pushing down. But the movement of the elevons and the command is not strong enough to make the turn,  When the plane starts to pitch down, the command in the elevons continues to be right turn and I don't see a strong pitch up command to compensate and hold the altitude. Then very shortly after that because of the speed we see the flutter...

tubeme, 2018-03-21:
I see where might be the problem...

Because of #9078 we had to make FW_P_LIM_MAX = 10

So we cut the possibility to get out of such outside envelope situations by default.

It is another question why the autopilot allowed the plane to get in the situation outside its predefined envelope.

So we are in a dead lock.

tubeme, 2018-03-21:
Could FW_P_LIM_MAX = 10 be the problem?

LorenzMeier, 2018-03-21:
Depending on how the airframe is tuned (controller gains) and the mixers are ordered its possible that it ran into either controller output or mixer saturation.

tubeme, 2018-03-21:
Thanks @LorenzMeier. We dedicated a lot of time on tuning and the plane is performing good to our taste.

The only thing that really changed since recently is the FW_P_LIM_MAX = 10 from  FW_P_LIM_MAX = 25.

We did this to fight #9078

We are really stuck now. We like the 1.7+ but we are afraid to make a mission with our expensive camera. But at the same time we don't want to go to 1.6.5 because there are a lot of mission related fixes in 1.7.3 we need,

Here is how the mixer looks like. We use only the main bus.

# VTOL quad X mixer for PX4FMU
#=============================

R: 4x 10000 10000 10000 0

Elevon mixers
-------------
Three scalers total (output, roll, pitch).

On the assumption that the two elevon servos are physically reversed, the pitch
input is inverted between the two servos.

M: 2
O:      10000  10000      0 -10000  10000
S: 1 0   7500   7500      0 -10000  10000
S: 1 1   6500   6500      0 -10000  10000

M: 2
O:       10000   10000    0 -10000  10000
S: 1 0    7500    7500    0 -10000  10000
S: 1 1   -6500   -6500    0 -10000  10000

Throttle mixer
--------------
M: 1
O:      10000  10000      0 -10000  10000
S: 1 3      0  20000 -10000 -10000  10000

Z:

Nothing unusual. A little more weight to the roll in the mixer.

JohnSnowball, 2018-03-22:
Hello, I checked your log and found that, the logger stopped (maybe the whole PX4) in air, just as mine. The firmware I'm using is derived from master in July 2017.

tubeme, 2018-03-22:
@JohnSnowball Thanks man. I think just the logging stopped because we were able to switch back to MC just before we hit the ground...

RomanBapst, 2018-04-30:
@tubeme I know this is abit late but do you happen to have the telemetry log of this flight?

tubeme, 2018-04-30:
Hmmm It will be really hard to find out this one. I will try.

From the link I gave

LOG: https://logs.px4.io/plot_app?log=89bd8fa6-6506-4e9a-87d5-81ee2cc9243a

I was able to download ULG file...

Other than that I will try to find it...

RomanBapst, 2018-05-03:
@tubeme Hi Vasil, do you know if the autopilot lost power during impact?

SinigerVTL, 2018-05-03:
@RomanBapst On impact the Batteries just flew away. It is seen in the video.

But up until the crash the vtol had power, the pilot switched the transition switch but too late. Quad-copter was working before the impact.

tubeme, 2018-05-03:
@RomanBapst On impact the Batteries just flew away. It is seen in the video.

But up until the crash the vtol had power, the pilot switched the transition switch but too late. Quad-copter was working before the impact.