PX4/PX4-Autopilot #9110: mRo ms5525 low airspeed

JonReacher, 2018-03-19:
Testing airspeed sensor on test platform. Zero wind, fresh calibration. Driving in car with test platform mounted outside in clean airflow. Pitot in front of platform in clean air. 

Indicated airspeed is consistently 8-10 mph low from ground speed and car speedometer. I've had some landing issues recently with too low airspeed resulting in stalls well before touching down. I'm new to 1.7.3 and the ms5525. On previous versions and with ms4525 I had no issues with airspeed being so far off. I was hoping the changes would improve things including calibration warnings, but it does not seem to have. I still get regular warnings as well.

dagar, 2018-03-19:
Do you have a flight log you could share?

In the car test how far was the pitot from the vehicle? Have you done this same test with other sensors? Which pitot tube are you using?

JonReacher, 2018-03-19:
Yes, log is here. https://logs.px4.io/plot_app?log=bbd78848-d3d0-4e12-a9cd-56ba3612ed53

5 mins is the turnaround so either side of that is 180 deg difference. Pitot was about 3 feet to the side of the car. I did this test with a ms4525 and it was very accurate. This was last year with older version. The problem with that was constant airspeed cal error warnings.

JonReacher, 2018-03-19:
I'm using the mRo pitot tube. The more 'squared' end, as opposed to 'pointy' pitots out there.

Antiheavy, 2018-03-20:
> the mRo pitot tube

have you checked to see if this particular pitot tube leaks?  I generally like mRo products, but I have not used their Pitot.  The mRo pitot looks suspiciously like the old 3DR pitot which was terribly leaky.  I can confidently recommend the eagle tree pitot (the original pitot of this style as far as I know) as I've used many 100's of them over the years and they rarely leak.

this is how we check that our pitot tubes are reading positive and do not leak.  press the soft part of our finger hard over the end of the pitot to create some pressure inside, hold for several seconds, you should see airspeed spike positive and then slowly climb (how much depends on how much tubing you have before your pressure sensor).
[image]

JonReacher, 2018-03-20:
I usually do the finger press for calibration, which I learned from a post of yours a while back -thanks! It does exactly as you say, so it seems good. Sometimes I get a calibration that sits at O even with a little wind, sometimes it jumps between 1-5ish mph. I wonder if that shows any indication of an off calibration. Is it even possible to do an off/ bad calibration, or is it pass/ fail only?

JonReacher, 2018-03-21:
Is it possible to disconnect the static tubing from the pitot and just leave it in the aircraft bay? I've tried doing this on the bench and it seems to work perfectly. Would this help prevent improper offsets by removing quite a bit of variability from the equation?

JonReacher, 2018-03-21:
Yeah, I can see how pressure changes could throw things off. I was thinking it would be static in there, but that's probably wishful thinking.

 I didn't know about the new firmware would allow dual sensor comparison. I'll have to set that up. I do keep seeing the eagle tree pitot being mentioned, as well as the SPD3x sensor.

Aside from ensuring actual sensor accuracy, I'm also trying to figure out how to eliminate all of the 'check airspeed cal or pitot warnings' that seem to happen on a good cal.

closed 2019-07-09 by stale[bot]