PX4/PX4-Autopilot #9114: mission yaw

Stifael, 2018-03-20:
Currently when doing a mission, the navigator adjusts the yaw such that the vehicle always points along the line previous-current setpoint. However, it even does that if the desired mission yaw is equal NAN, which is defined as having a desired yaw unchanged to the previous one (please see at MAV_CMD http://mavlink.org/messages/common#ATT_POS_MOCAP: `Desired yaw angle at waypoint (rotary wing). NaN for unchanged)`. 
To me, unchanged means that yaw is not important and hence, in case of a MC, the vehicle should not rotate at all. 
I would like to get rid of the heading update in mission (basically get rid of https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission.cpp#L224) and just send out the yaw waypoint received from the uploaded mission. If the yaw waypoint is NAN, then the vehicle should not rotate at all. If there is a desired yaw at a waypoint, then rotate into that yaw while flying and keep it until the waypoint is reached (not sure about that: maybe there is something more intuitive). 
One other thing that I am not sure about: there should be a way to tell the vehicle that it should point along the path during survey.

Stifael, 2018-03-20:
I also would like know the definition of MAV_CMD_NAV_LOITER_TIME param4, which can have three different meanings. In px4, it checks for exit track location. However, in qgc I don't see that option. I see that there is a yaw setpoint option though

Stifael, 2018-03-20:
@dagar are you sure that https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_mission.cpp#L1310 has to be converted to radians? I think it is already in radians? 
In addition, for multicopter I think this is never used, because the vehicle anyway always rotates toward the current waypoint.

dagar, 2018-03-20:
No not sure at all. As far as I can tell it falls outside of the spec. Given that it's typically user facing editable in QGC planning I think it's probably intended to be degrees.

This is the type of ambiguity that needs to be removed from the Mavlink spec.

dagar, 2018-03-20:
I think getting the yaw update out of master is definitely the way to go. This is both multicopter specific, and easier to handle in the multicopter controller.

I was also hoping to move the FOH altitude behaviour to the FW controller in https://github.com/PX4/Firmware/pull/8883. 

After that we can be rid of the continuous navigator position setpoint triplet publication.

mlaiacker, 2019-05-17:
Hi, there is a double radian conversion for all yaw settings in
https://github.com/PX4/Firmware/blob/d78a842ca802396c4d0d88ca6d17c9ddd0e4d770/src/modules/mavlink/mavlink_mission.cpp#L1328
QGC is sending param4 as radians. 

It is not defined in the mavlink spec: "Desired yaw angle."

So all the "math::radians" have to be removed or the implementation in QGC changed. 

I came across this Issue because I want to modify the loiter behavior when a yaw angle is defined.
Any solutions to this Issue? 
I fixed it in my branch but this should be fixed in the mavlink spec and the PX4 implementation. 
How is it implemented in Ardupilot?

FlavioTonelli, 2019-05-17:
@mlaiacker Which is your QGC version? 
Last QGC shall already send yaw angle as degree.

mlaiacker, 2019-05-20:
Oh ok, I used 3.5.0

bresch, 2019-07-10:
@Stifael What's the status on that?