PX4/PX4-Autopilot #9116: PX4 firmware compilation fails without mc_pos_control module

philipoe, 2018-03-20:
**Compilation issue**

#### Issue: 
The PX4 firmware compilation (current master, d87b7ac) fails if one does NOT compile in the `mc_pos_control` module. For rovers, fixed-wings, etc. one might not want to to compile that in to save flash space, so many users might not compile that in (for examle: our team). It fails due to unknown parameters [here](https://github.com/PX4/Firmware/blob/master/src/modules/navigator/precland.h#L130) .

#### Solution:
@ndepal: Can you make your implementation more agnostic to the vehicle type?

ndepal, 2018-03-20:
@bkueng Is this a consequence of the new param class that the absence of a parameter now is a compile-time error?

Since `precland` can't be used for non-multicopter vehicles it would be fine if the parameter was not found at runtime since it would never run ayway.

Have you had this problem in other places?

bkueng, 2018-03-20:
Yes it's a consequence of the new param class: it guarantees that every parameter that could potentially be accessed at runtime exists in the build. I think this is highly desirable as it increases the stability. It already uncovered some broken dependencies (https://github.com/PX4/Firmware/pull/8789/commits/70db3d2c5582e302a7bd638f28d823870d65dac3).

We can do 2 things:
- Avoid using parameters from other modules. This is not always possible, but will improve modularity and reduce complexity.
- Change the way how parameter scoping is done to have more control over it. For example a module could specify which param (or param group) it depends on. Or the other way around: a param (group) specifies a list of modules where it's required.

@philipoe short-term you can disable the param scoping like it's done here: https://github.com/PX4/Firmware/blob/master/cmake/configs/qurt_sdflight_default.cmake#L25. This will add all params to the build and thus increase the binary size a bit.

dagar, 2018-03-20:
Let's get an issue going to track the known problems. One relatively easy solution that fits with the current architecture is to have the relevant module maintain exclusive ownership of the parameter and publish anything needed by navigator, etc.

For example the FW position controller publishes the waypoint acceptance radius used by navigator. In the VTOL/navigator case that also works well. I haven't looked at precision landing yet.

sandalski, 2018-04-25:
I am not sure if this is related or if I should open a new issue but trying to build the latest firmware without `modules/vtol_att_control` fails because of missing parameters. (e.g. error: 'VT_B_DEC_MSS' is not a member of 'px4::params'). Without removing vtol_att_control it's very difficult to get uavcan module to fit in a pixhawk 1 firmware image.

dagar, 2019-02-11:
This has been resolved.

closed 2019-02-11 by dagar