PX4/PX4-Autopilot #9118: **Feature Request / Discussion**: Airframe group definitions for Planes

philipoe, 2018-03-20:
**Feature Request / Discussion**

The [PX4/QGC airframes ](https://dev.px4.io/en/airframes/airframe_reference.html) for planes are in my opinion somewhat ambiguously defined. The obviously good concept of grouping airframes was merged (and most likely also discussed) [here](https://github.com/PX4/Firmware/pull/7482).

Currently, there is the groups "standard plane" and "Plane A-Tail", i.e. we group them by tail configuration. However, the planes in these groups also seem to differ by their flap configuration (Plane A-Tail: Two separate flaps, Standard Plane: Flaps on same servo output or not even connected).

Overall Question: Why were the groups chosen that way? I assume just because at that time there was only a limited amount of platforms. Overall however, I could think of the airframe types:
 - Plane A/V-Tail, with 2 flaps
 - Plane A/V-Tail, with 0/1 flaps
 - Plane T-Tail, with 2 flaps - Plane T-Tail, with 2 flaps
 - Would it instead makes sense to either a) group this according to flap configuration instead of tail configuration?
 - In the end, the problem is that 

http://discuss.px4.io/

hamishwillee, 2019-01-29:
@philipoe Sorry, completely missed this issue. Generally the design has been based around reducing the selection as far as possible - to a set from which a user can reasonably make a selection that will work (ie previously we have millions of specific tunings that were effectively the same).
Currently the set looks like this: 
[image]

Anyway, sounds like you may have a point. @dagar @RomanBapst  ?

hamishwillee, 2019-06-27:
Keep this open