PX4/PX4-Autopilot #9120: QGroundControl daily: No airspeed sensor calibration available

santiago3dr, 2018-03-20:
Flashed Beta 1.7.4 and i am not given the option to calibrate airspeed sensor
[image]

Was able to successfully calibrate all other sensors

Airspeed calibration looks to be available on qgroundcontrol stable 3.3.0
[image]

Running a Pixhawk mini (v3 build) on a Phantom Flying Wing

dagar, 2018-03-27:
See https://github.com/mavlink/qgroundcontrol/pull/6270

dagar, 2018-03-27:
@santiago3dr this is fixed in QGC daily. https://github.com/mavlink/qgroundcontrol/pull/6272

closed 2018-03-27 by dagar

santiago3dr, 2018-03-27:
great!
thanks