PX4/PX4-Autopilot #9138: TypeError: norm() got an unexpected keyword argument 'axis'

Jai-GAY, 2018-03-22:
reference from [here](https://github.com/PX4/Firmware/issues/9124) , anyone got the same error running this command [make px4fmu-v2_default](https://dev.px4.io/en/setup/building_px4.html) ?

[4/158] Generating mixer_multirotor.generated.h
FAILED: cd /home/user/px4/github/Firmware/build/px4io-v2_default/src/lib/mixer && /usr/bin/python /home/user/px4/github/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py --normalize -f /home/user/px4/github/Firmware/src/lib/mixer/geometries/dodeca_bottom_cox.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/dodeca_top_cox.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/hex_cox.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/hex_plus.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/hex_t.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/hex_x.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/octa_cox.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/octa_cox_wide.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/octa_plus.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/octa_x.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_deadcat.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_h.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_plus.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_s250aq.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_vtail.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_wide.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_x.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_x_pusher.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_y.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/tri_y.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/twin_engine.toml -o mixer_multirotor_normalized.generated.h
Traceback (most recent call last):
  File "/home/user/px4/github/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py", line 336, in <module>
    A, B = geometry_to_mix(geometry)
  File "/home/user/px4/github/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py", line 182, in geometry_to_mix
    At = geometry_to_thrust_matrix(geometry)
  File "/home/user/px4/github/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py", line 166, in geometry_to_thrust_matrix
    Ct=np.array([[rotor['Ct']] for rotor in geometry['rotors']])).T
  File "/home/user/px4/github/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py", line 155, in thrust_matrix
    ax = axis / np.linalg.norm(axis, axis=1)[:, np.newaxis]
TypeError: norm() got an unexpected keyword argument 'axis'
ninja: build stopped: subcommand failed.
make[1]: *** [px4io-v2_default] Error 1

Jai-GAY, 2018-03-22:
anyone thinks is due to [numpy version](https://github.com/nipy/nipype/issues/1104) ?

tstellanova, 2018-09-01:
I encountered this same problem and in my case I believe it was caused by a bad `numpy` and `pandas` version.  I had to reinstall `sudo -H pip install pandas`-- after that I was able to build px4fmu-v5_default

dagar, 2018-09-01:
I've been gathering up a requirements.txt here. https://github.com/PX4/Firmware/blob/master/Tools/setup/requirements.txt

Once it's been tested we can point the official documentation and docker builds at it.

closed 2019-07-08 by stale[bot]