PX4/PX4-Autopilot #9139: Suspectible servo broken and PX4 dead

JohnSnowball, 2018-03-22:
Hello, here is a standard vtol, I made successful flight for about 10 times this month, however yesterday it crashed in a strange way. Here's the log:
https://logs.px4.io/plot_app?log=89a9aee2-cb5d-4451-b816-925f508ebd64
PX4FMU-V2， no modification in flight control modules from master.

All begins normally, and during mission to the third waypoint, it starts sinking with a very big negative roll angle. The after a few circles, I switched to the RTL mode, it continues to sink! Then when it starts to nose dive, I tried to turn the plane into multicopter mode, no reaction, crash. Please see the 3D replay, it could be more clear for you.

I checked log, found logger stopped when plane is still in air(start to nose dive), I guess that's why it didn't do back transition.

I got two probelems:
**1. Why can't the plane reach it's desired roll angle suddenly?** Is it due to a servo broken? How to confirm this point? I think it's the biggest problem that made plane sink.
[image]

**2. Why PX4 is dead?** I have no idea about this. Is it a hardware problem? Or stack overflow?

If you need more information, I will try my best to support you, but I can't fly this plane anymore as it's totally crashed...

JohnSnowball, 2018-03-22:
@RomanBapst suppose you are an expert in vtol...

mhkabir, 2018-03-22:
Possibly related : https://github.com/PX4/Firmware/issues/9108

bkueng, 2018-03-28:
@JohnSnowball do you remember if you made any parameter changes or did a calibration before the flight (without rebooting)?
Do you have a QGC telemetry log?

JohnSnowball, 2018-04-08:
@bkueng I really did some parameter changes, but maybe no calibration as I have made about 10 flights before for the same plane. How to get that QGC telementry log?

bkueng, 2018-04-09:
> How to get that QGC telementry log?

They are stored under the 'File Save Path': https://docs.qgroundcontrol.com/en/SettingsView/General.html#miscellaneous