PX4/PX4-Autopilot #9145: ERROR when make posix_sitl_default gazebo

goodman110110110, 2018-03-23:
Hello, my computer is ubuntu 16.04, gazebo 7 and ros kientic.
All works ok but then i installed cartographer.

when I make posix_sitl_default gazebo, it comes some errors as follows:

ghf@ghf-A8SR:~/pixhawk1.7.3/Firmware$ make posix_sitl_default gazebo
-- Build Type: RelWithDebInfo
-- PX4 VERSION: v1.7.3
-- CONFIG: posix_sitl_default
-- Build Type: RelWithDebInfo
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- C compiler: cc (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609
-- C++ compiler: c++ (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ghf/pixhawk1.7.3/Firmware/build/posix_sitl_default
[161/572] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   thread
--   signals
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
--   chrono
[... 119 lines cut ...]
       ::gazebo::msgs::Quaternion::internal_default_instance());
       ^
SensorImu.pb.cc:111:7: error: ‘internal_default_instance’ is not a member of ‘gazebo::msgs::Vector3d’
       ::gazebo::msgs::Vector3d::internal_default_instance());
       ^
SensorImu.pb.cc:113:7: error: ‘internal_default_instance’ is not a member of ‘gazebo::msgs::Vector3d’
       ::gazebo::msgs::Vector3d::internal_default_instance());
       ^
SensorImu.pb.cc: In function ‘void sensor_msgs::msgs::protobuf_SensorImu_2eproto::{anonymous}::AddDescriptorsImpl()’:
SensorImu.pb.cc:138:19: error: ‘gazebo::msgs::protobuf_quaternion_2eproto’ has not been declared
   ::gazebo::msgs::protobuf_quaternion_2eproto::AddDescriptors();
                   ^
SensorImu.pb.cc:139:19: error: ‘gazebo::msgs::protobuf_vector3d_2eproto’ has not been declared
   ::gazebo::msgs::protobuf_vector3d_2eproto::AddDescriptors();
                   ^
ninja: build stopped: subcommand failed.
[568/572] Linking CXX executable px4
ninja: build stopped: subcommand failed.
Makefile:147: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 1

I find in the google [that](http://answers.gazebosim.org/question/18096/this-file-was-generated-by-an-older-version-of-protoc/) the "Gazebo is built against the version of protoc released to the ubuntu repos. On 16.04 that version is 2.6.1." but cartographer need 3.4.1.

How can I use gazebo and cartographer all together. How to solve the problem?
change the version of protobuf?

dagar, 2018-03-23:
I'm not sure how to rectify the version difference. Could you try building newer gazebo with newer protobufs?

goodman110110110, 2018-03-24:
@dagar I have not try build new gazebo with newer protobufs. My gazebo 7 is default of ROS kientic. How to update gazebo?

lamping7, 2018-03-24:
Google is your friend... https://github.com/googlecartographer/cartographer_ros/issues/625
http://answers.gazebosim.org/question/18096/this-file-was-generated-by-an-older-version-of-protoc/

closed 2018-03-24 by lamping7