PX4/PX4-Autopilot #9150: MultiRotor: Unstable manual flight

santiago3dr, 2018-03-23:
Flash current master d2712dc and took off in stabilized; very jumpy/jittery; slow oscillations in heading
Running Pixracer (V4)
https://logs.px4.io/plot_app?log=6ca0b878-b8a4-4700-8040-e09083db4c04

re-calibrated and tried again; same result:
https://logs.px4.io/plot_app?log=64eac19d-2e42-4450-85e3-39dd31930393

compared to stable flight with 1.7.4 beta:
https://logs.px4.io/plot_app?log=12136a55-46ff-449a-bff3-38c58b832542

@dagar this might be related to recent changes to motor saturation handling?

santiago3dr, 2018-03-23:
Flew with last version of master prior to Multicopter Mixer changes ( 01f5f88 )
Copter flew very stable:
https://logs.px4.io/plot_app?log=a89662aa-4609-4a91-b623-60ee1cc696bb

dagar, 2018-03-24:
MC_AIRMODE was set 0.

dagar, 2018-03-24:
Link to PR for reference. https://github.com/PX4/Firmware/pull/9062

RomanBapst, 2018-03-26:
I agree with @bresch - IMU_GYRO_CUTOFF was set to 80 Hz but MC_DTERM_CUTOFF is set to 0, this will inject too much noise into the D path of the PID controller.

MaEtUgR, 2018-03-26:
@bresch @RomanBapst I agree with both of you, why was the default D term cutoff changed in the airmode pr?
https://github.com/bresch/Firmware/commit/9332881ff98981d6a39fe0327e0f63f8f9dc3f07#diff-683f0a3184d233b6b7f6e84a05d98f31R545
It looks like a rebase error to me because #9070 was merged first and the D term cutoff was reverted...
Pr with a hot-fix: #9162