PX4/PX4-Autopilot #9151: Internal HMC5883 not (always) starting on FMUv2 (Original 3DR Pixhawk)

ryanjAA, 2018-03-24:
Internal HMC5883 doesn't always start on boot and can not start from shell. Must reboot a few ties to get it enabled. HUD/artificial horizon and mag are frozen when not working and some other functionality also doesn't work when this occurs. (It's not hardware related since going back a few versions fixes it). Mixer freezes up too on the ailerons. Flaps don't respond correctly, to be clear that means flaps don't do anything and ailerons go full when using the flap switch (mixer I'm using is for the Albatross AAVVTWFF which ties the flaps and ailerons together when hitting the flap button but also tried it on a custom mixer here that is similar but with a few other inputs, same thing). Telemetry does work over Telem1. Telem2 TBD, had issues earlier but could be unrelated. This is very similar to issue [7327](https://github.com/PX4/Firmware/issues/7327) except this is an Original 3DR Pixhawk and it is the HMC5883 not starting.

ryanjAA, 2018-03-24:
[image]

dagar, 2018-03-25:
Possibly already fixed in master by https://github.com/PX4/Firmware/pull/9074.

ryanjAA, 2018-03-26:
Tentatively can confirm that master fixes this. Have checked 1x board. Will run through several variants to finalize.

ryanjAA, 2018-05-13:
Fixed.

closed 2018-05-13 by ryanjAA