PX4/PX4-Autopilot #9153: unused args [respawn_gazebo] for include of [/opt/ros/indigo/share/gazebo_ros/launch/empty_world.launch]

varunvp, 2018-03-24:
**Bug Report** or **Feature Request**

When I run the following commands, as on the website https://dev.px4.io/en/simulation/ros_interface.html,
```
cd <Firmware_clone>
make posix_sitl_default gazebo
source ~/catkin_ws/devel/setup.bash    // (optional)
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/posix_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 posix_sitl.launch
```
 I get this error when executing the last line
```
unused args [respawn_gazebo] for include of [/opt/ros/indigo/share/gazebo_ros/launch/empty_world.launch]
The traceback for the exception was written to the log file
```

This is what is there in the log file

```
[roslaunch][INFO] 2018-03-24 15:44:10,058: Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2018-03-24 15:44:10,060: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2018-03-24 15:44:10,060: roslaunch starting with args ['/opt/ros/indigo/bin/roslaunch', 'px4', 'posix_sitl.launch']
[roslaunch][INFO] 2018-03-24 15:44:10,060: roslaunch env is {'MANDATORY_PATH': '/usr/share/gconf/ubuntu.mandatory.path', 'ROS_DISTRO': 'indigo', 'ROS_LOG_FILENAME': '/home/varun/.ros/log/17ed39e8-2f4c-11e8-8821-f48e38f33bbf/roslaunch-varun-5577-8808.log', 'XDG_GREETER_DATA_DIR': '/var/lib/lightdm-data/varun', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', 'XDG_CURRENT_DESKTOP': 'Unity', 'UPSTART_EVENTS': 'started starting', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'PKG_CONFIG_PATH': '/home/varun/catkin_ws/devel/lib/pkgconfig:/opt/ros/indigo/lib/pkgconfig', 'ROSLISP_PACKAGE_DIRECTORIES': '/home/varun/catkin_ws/devel/share/common-lisp', 'CPATH': '/home/varun/catkin_ws/devel/include:/opt/ros/indigo/include', 'LOGNAME': 'varun', 'USER': 'varun', 'PATH': '/home/varun/catkin_ws/devel/bin:/opt/ros/indigo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games', 'XDG_VTNR': '7', 'GNOME_KEYRING_CONTROL': '/run/user/1000/keyring-C7ZQLH', 'CMAKE_PREFIX_PATH': '/home/varun/catkin_ws/devel:/opt/ros/indigo', 'LD_LIBRARY_PATH': '/home/varun/catkin_ws/devel/lib:/opt/ros/indigo/lib:/home/varun/src/Firmware/Tools/sitl_gazebo/Build/msgs/:/home/varun/src/Firmware/build/posix_sitl_default/build_gazebo:/home/varun/src/Firmware/Tools/sitl_gazebo/Build/msgs/:/home/varun/src/Firmware/build/posix_sitl_default/build_gazebo', 'LANG': 'en_IN', 'TERM': 'xterm', 'SHELL': '/bin/bash', 'XDG_SESSION_PATH': '/org/freedesktop/DisplayManager/Session0', 'XAUTHORITY': '/home/varun/.Xauthority', 'LANGUAGE': 'en_IN:en', 'COMPIZ_CONFIG_PROFILE': 'ubuntu', 'SHLVL': '1', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'QT_QPA_PLATFORMTHEME': 'appmenu-qt5', 'JOB': 'gnome-session', 'TEXTDOMAIN': 'im-config', 'UPSTART_INSTANCE': '', 'CLUTTER_IM_MODULE': 'xim', 'WINDOWID': '60817420', 'SESSIONTYPE': 'gnome-session', 'XMODIFIERS': '@im=ibus', 'ROS_ETC_DIR': '/opt/ros/indigo/etc/ros', 'GPG_AGENT_INFO': '/run/user/1000/keyring-C7ZQLH/gpg:0:1', 'HOME': '/home/varun', 'QT4_IM_MODULE': 'xim', 'SELINUX_INIT': 'YES', 'XDG_RUNTIME_DIR': '/run/user/1000', 'INSTANCE': 'Unity', 'PYTHONPATH': '/home/varun/catkin_ws/devel/lib/python2.7/dist-packages:/opt/ros/indigo/lib/python2.7/dist-packages', 'SSH_AUTH_SOCK': '/run/user/1000/keyring-C7ZQLH/ssh', 'VTE_VERSION': '3409', 'ROS_ROOT': '/opt/ros/indigo/share/ros', 'GDMSESSION': 'ubuntu', 'IM_CONFIG_PHASE': '1', 'UPSTART_JOB': 'unity-settings-daemon', 'TEXTDOMAINDIR': '/usr/share/locale/', 'GNOME_KEYRING_PID': '1467', 'XDG_SEAT_PATH': '/org/freedesktop/DisplayManager/Seat0', 'ROS_PACKAGE_PATH': '/home/varun/catkin_ws/src/hector_models/hector_components_description:/home/varun/catkin_ws/src/hector_gazebo/hector_gazebo:/home/varun/catkin_ws/src/hector_gazebo/hector_gazebo_plugins:/home/varun/catkin_ws/src/hector_gazebo/hector_gazebo_thermal_camera:/home/varun/catkin_ws/src/hector_gazebo/hector_gazebo_worlds:/home/varun/catkin_ws/src/hector_localization/hector_localization:/home/varun/catkin_ws/src/hector_models/hector_models:/home/varun/catkin_ws/src/hector_localization/hector_pose_estimation_core:/home/varun/catkin_ws/src/hector_localization/hector_pose_estimation:/home/varun/catkin_ws/src/hector_quadrotor/hector_quadrotor:/home/varun/catkin_ws/src/hector_quadrotor/hector_quadrotor_demo:/home/varun/catkin_ws/src/hector_quadrotor/hector_quadrotor_description:/home/varun/catkin_ws/src/hector_models/hector_sensors_description:/home/varun/catkin_ws/src/hector_gazebo/hector_sensors_gazebo:/home/varun/catkin_ws/src/hector_quadrotor/hector_uav_msgs:/home/varun/catkin_ws/src/hector_quadrotor/hector_quadrotor_controller:/home/varun/catkin_ws/src/hector_quadrotor/hector_quadrotor_controller_gazebo:/home/varun/catkin_ws/src/hector_quadrotor/hector_quadrotor_model:/home/varun/catkin_ws/src/hector_quadrotor/hector_quadrotor_gazebo_plugins:/home/varun/catkin_ws/src/hector_quadrotor/hector_quadrotor_pose_estimation:/home/varun/catkin_ws/src/hector_quadrotor/hector_quadrotor_teleop:/home/varun/catkin_ws/src/hector_models/hector_xacro_tools:/home/varun/catkin_ws/src/mav_comm/mav_comm:/home/varun/catkin_ws/src/mav_comm/mav_msgs:/home/varun/catkin_ws/src/mavlink:/home/varun/catkin_ws/src/mavros/libmavconn:/home/varun/catkin_ws/src/mavros/mavros_msgs:/home/varun/catkin_ws/src/mavros/mavros:/home/varun/catkin_ws/src/mavros/mavros_extras:/home/varun/catkin_ws/src/hector_localization/message_to_tf:/home/varun/catkin_ws/src/hector_quadrotor/hector_quadrotor_gazebo:/home/varun/catkin_ws/src/mav_comm/planning_msgs:/home/varun/catkin_ws/src/rotors_simulator/rotors_comm:/home/varun/catkin_ws/src/rotors_simulator/rotors_control:/home/varun/catkin_ws/src/rotors_simulator/rotors_description:/home/varun/catkin_ws/src/rotors_simulator/rotors_evaluation:/home/varun/catkin_ws/src/rotors_simulator/rotors_gazebo_plugins:/home/varun/catkin_ws/src/rotors_simulator/rotors_gazebo:/home/varun/catkin_ws/src/rotors_simulator/rotors_hil_interface:/home/varun/catkin_ws/src/rotors_simulator/rotors_joy_interface:/home/varun/catkin_ws/src/rotors_simulator/rotors_simulator:/home/varun/catkin_ws/src/rotors_simulator/rqt_rotors:/home/varun/catkin_ws/src/mavros/test_mavros:/opt/ros/indigo/share:/opt/ros/indigo/stacks:/home/varun/src/Firmware:/home/varun/src/Firmware/Tools/sitl_gazebo', 'GAZEBO_MODEL_PATH': ':/home/varun/src/Firmware/Tools/sitl_gazebo/models:/home/varun/src/Firmware/Tools/sitl_gazebo/models', 'XDG_SESSION_ID': 'c2', 'DBUS_SESSION_BUS_ADDRESS': 'unix:abstract=/tmp/dbus-LzsK7glPDG', '_': '/opt/ros/indigo/bin/roslaunch', 'DEFAULTS_PATH': '/usr/share/gconf/ubuntu.default.path', 'GTK_IM_MODULE': 'ibus', 'DESKTOP_SESSION': 'ubuntu', 'UPSTART_SESSION': 'unix:abstract=/com/ubuntu/upstart-session/1000/1470', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-ubuntu:/usr/share/upstart/xdg:/etc/xdg', 'GTK_MODULES': 'overlay-scrollbar:unity-gtk-module', 'OLDPWD': '/usr/local/bin', 'GDM_LANG': 'en_US', 'GAZEBO_PLUGIN_PATH': '/home/varun/src/Firmware/build/posix_sitl_default/build_gazebo:/home/varun/src/Firmware/build/posix_sitl_default/build_gazebo:', 'XDG_DATA_DIRS': '/usr/share/ubuntu:/usr/share/gnome:/usr/local/share/:/usr/share/', 'PWD': '/home/varun/src/Firmware', 'QT_IM_MODULE': 'ibus', 'ROS_MASTER_URI': 'http://localhost:11311', 'COLORTERM': 'gnome-terminal', 'DISPLAY': ':0', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arj=01;31:*.taz=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lz=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.axv=01;35:*.anx=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36: [... cut ...]
RLException: unused args [respawn_gazebo] for include of [/opt/ros/indigo/share/gazebo_ros/launch/empty_world.launch]

[rospy.core][INFO] 2018-03-24 15:44:10,217: signal_shutdown [atexit]
```

In the posix_sitl.launch file, I found the unused argument
`        <arg name="respawn_gazebo" value="$(arg respawn_gazebo)"/>`

When I removed this line and run the simulation, it worked as expected. I want to know if this is a bug, or I'm missing something.

I'm running ROS Indigo on Ubuntu 14.04, using Gazebo 7.

lamping7, 2018-03-24:
First, the dev guide states ROS Kinetic is the supported version, along with Ubuntu 16.04. 

Second, the gazebo argument exists here:
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/9cbc4bfd9b0caf763387cb203f6bf695071cae07/gazebo_ros/launch/empty_world.launch#L17

So, either you have version issues, or you modified the `posix_sitl.launch` file, in a way where it is no longer defining the argument. i.e. you must have this line: https://github.com/PX4/Firmware/blob/ed261c76daf6e5a40fab7cec6079b1febff4be85/launch/posix_sitl.launch#L23
to accompany this line:
https://github.com/PX4/Firmware/blob/ed261c76daf6e5a40fab7cec6079b1febff4be85/launch/posix_sitl.launch#L33

You'd have to post your complete version of the launch file for me to see this. I suggest restoring the launch file provided by PX4 to it's unmodified state and use the recommended versions. I'm closing this, as there is no bug, but fell free to post back if you still have issues after following the suggested actions.

closed 2018-03-24 by lamping7

varunvp, 2018-03-24:
I compared your launch file with the launch file in my gazebo_ros directory, and mine doesn't have respawn_gazebo argument. As for the launch file, here it is, as is

```
<?xml version="1.0"?>
<launch>
    <!-- Posix SITL environment launch script -->
    <!-- launches PX4 SITL, Gazebo environment, and spawns vehicle -->
    <!-- vehicle pose -->
    <arg name="x" default="0"/>
    <arg name="y" default="0"/>
    <arg name="z" default="0"/>
    <arg name="R" default="0"/>
    <arg name="P" default="0"/>
    <arg name="Y" default="0"/>
    <!-- vehicle model and world -->
    <arg name="est" default="ekf2"/>
    <arg name="vehicle" default="iris"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/$(arg vehicle)/$(arg vehicle).sdf"/>
    <arg name="rcS" default="$(find px4)/posix-configs/SITL/init/$(arg est)/$(arg vehicle)"/>
    <!-- gazebo configs -->
    <arg name="gui" default="true"/>
    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>
    <arg name="respawn_gazebo" default="false"/>
    <!-- PX4 SITL -->
    <node name="sitl" pkg="px4" type="px4" output="screen" args="$(find px4) $(arg rcS)"/>
    <!-- Gazebo sim -->
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="gui" value="$(arg gui)"/>
        <arg name="world_name" value="$(arg world)"/>
        <arg name="debug" value="$(arg debug)"/>
        <arg name="verbose" value="$(arg verbose)"/>
        <arg name="paused" value="$(arg paused)"/>
        <arg name="respawn_gazebo" value="$(arg respawn_gazebo)"/>    //**This is the  'unused' arg. If this is commented, it works properly**
    </include>
    <!-- gazebo model -->
    <node name="$(anon vehicle_spawn)" pkg="gazebo_ros" type="spawn_model" output="screen" args="-sdf -file $(arg sdf) -model $(arg vehicle) -x $(arg x) -y $(arg y) -z $(arg z) -R $(arg R) -P $(arg P) -Y $(arg Y)"/>
</launch>
```

I haven't done any modifications to this file. So this must likely be a version issue?

lamping7, 2018-03-24:
Yup, there's your problem... The argument was added to the gazebo_ros package with Kinetic back in 2015. I guess it didn't make it to Indigo.

The commit is here: https://github.com/ros-simulation/gazebo_ros_pkgs/commit/71dcc776c6efdd9408e69e3a39dcd55212419226#diff-a1a63a469cbe17191421efbe801f4491

So, either pull it out of the launch, like you did, or upgrade.

shriram75, 2020-06-17:
I am getting this error "unused args [interactive] for include of [/home/shriram/catkin_ws/src/Firmware/launch/posix_sitl.launch]
The traceback for the exception was written to the log file" when i run roslaunch command
This is my posix_sitl launch file.please tell me what i have to do
<launch>

    
    <arg name="x" default="0"/>
    <arg name="y" default="0"/>
    <arg name="z" default="0"/>
    <arg name="R" default="0"/>
    <arg name="P" default="0"/>
    <arg name="Y" default="0"/>
    <arg name="est" default="ekf2"/>
    <arg name="vehicle" default="iris"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/$(arg vehicle)/$(arg vehicle).sdf"/>
    <arg name="rcS" default="$(find px4)/posix-configs/SITL/init/$(arg est)/$(arg vehicle)"/>

    <arg name="headless" default="false"/>
    <arg name="gui" default="true"/>
    <arg name="ns" default="/"/>

    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>
   

    <node name="sitl" pkg="px4" type="px4" output="screen"
        args="$(find px4) $(arg rcS)">
    </node>

    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="headless" value="$(arg headless)"/>
        <arg name="gui" value="$(arg gui)"/>
        <arg name="world_name" value="$(arg world)" />
        <arg name="debug" value="$(arg debug)" />
        <arg name="verbose" value="$(arg verbose)" />
        <arg name="paused" value="$(arg paused)" />
        
    </include>

    <node name="$(anon vehicle_spawn)" output="screen" pkg="gazebo_ros" type="spawn_model"
        args="-sdf -file $(arg sdf) -model $(arg vehicle) -x $(arg x) -y $(arg y) -z $(arg z) -R $(arg R) -P $(arg P) -Y $(arg Y)"/>

</launch>

Jaeyoung-Lim, 2020-06-17:
@shriram75 Your issue is unrelated to this thread. Please open a new issue