PX4/PX4-Autopilot #9155: System is arming (or thinks it is arming) even with Critical warnings

Antiheavy, 2018-03-25:
PX4 1.7.3, installed on a Pixracer, using QGC v3.3.0.  The pixracer is sitting on my desk connected to QGC over USB.  I have a valid GPS fix.  I try to Arm and it seems to let me even with the following Critical warnings:

[image]

Is this right?

Antiheavy, 2018-03-25:
however, once i fixed the compass rotation problem (that was giving the "MAG SENSORS INCONSISTENT" warning), then the system properly prevents arming:
[image]

dagar, 2018-03-26:
Working on a fix here - https://github.com/PX4/Firmware/tree/pr-commander_preflight

taileron, 2018-03-27:
I think it´s not so bad to allow arming for stabilized flights and reject only e.g. auto mission or rtl that is directly dependant to the sensor´s errors

dagar, 2018-03-27:
@taileron I agree, but it needs to be done explicitly and I'm not entirely sure how we'd want to go about that.

Some of these requirements we could determine from the system configuration. For example if the user has and RC or data link loss failsafe configured for RTL then at arming time the system could require a valid global position estimate and valid home position.

taileron, 2018-03-27:
@dagar some errors disappear when the fc runs a minute but mostly it is necessary to reboot the system to get rid of conditions that could prevent arming.
so what´s about to have the possibility or do it automatic: repeat the preflight check after 40 ... 120 sec when temperatures and reference voltages are more stable and more gps satellites than directly after boot, deleting all previous preflight errors + nsh "commander check" with current sensor values

commit 81fdd045 (2018-03-27): [WIP] commander fix preflight checks blocking arming

 - fixes #9155
commit 444d5f35 (2018-03-27): commander cleanup prearm and preflight calls

 - fixes #9155
commit 4ee375cc (2018-03-30): commander state machine fix preflight and prearm error

 -  fixes #9155
commit a33c31fd (2018-03-30): commander state machine fix preflight and prearm error

 -  fixes #9155
closed 2018-04-17 by dagar